ARMmbed/mbed-os #3359: WIZwiki7500(ECO) no export possible from online ice

sstaub, 2016-12-02:
### Description
- Type: Bug
- Priority: Blocker
---------------------------------------------------------------
## Bug

**Target**
WIZwiki7500(ECO)

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**
online ide

An export from online-ide is not possible, no uVision4/5 gcc ...
In former times it was possible to export to a simple .zip file,
these option is also missing.

0xc0170, 2016-12-02:
cc @theotherjimmy

theotherjimmy, 2016-12-02:
>An export from online-ide is not possible, no uVision4/5 gcc ...

When you say not possible, do you mean that the option is greyed out? Or do you mean something else?

My investigation shows that `make_armc5` is the only exporter that will work for the WIZwiki_7500.

>In former times it was possible to export to a simple .zip file, these option is also missing.

You could export to a zip file, but the behavior of that zip file was the same as `mbed import` and has much worse performance.

sstaub, 2016-12-02:
No, the options are all available: 
[image]
but export aborts:
[image]

theotherjimmy, 2017-01-03:
@sstaub 
The options being available is a bug with the website that should be fixed in the near future.

The export aborting is related to the fact that your target does not support the UVision5 export (UVision 4 is deprecated in favor of 5). The only export that your target supports is make_armc5.

sg-, 2017-02-03:
@sstaub @theotherjimmy Has this been resolved?

sstaub, 2017-02-04:
No, the only thing that changed is the selection (GCC) but the export itself doesn't work. I need export to uVision5.

theotherjimmy, 2017-05-16:
@sstaub It could be supported, but we would need to know the part name from CMSIS PACKs in the `device_name` key in targets.json.

sstaub, 2017-06-01:
@theotherjimmy I didn't really know what you mean. There are three boards
in targets.json:
`
    "WIZWIKI_W7500": {
        "supported_form_factors": ["ARDUINO"],
        "core": "Cortex-M0",
        "extra_labels": ["WIZNET", "W7500x", "WIZwiki_W7500"],
        "supported_toolchains": ["uARM", "ARM", "GCC_ARM", "IAR"],
        "inherits": ["Target"],
        "device_has": ["ANALOGIN", "I2C", "INTERRUPTIN", "PORTIN", "PORTINOUT", "PORTOUT", "PWMOUT", "RTC", "SERIAL", "SPI", "SPISLAVE", "STDIO_MESSAGES"],
        "release_versions": ["2", "5"]
    },
    "WIZWIKI_W7500P": {
        "supported_form_factors": ["ARDUINO"],
        "core": "Cortex-M0",
        "extra_labels": ["WIZNET", "W7500x", "WIZwiki_W7500P"],
        "supported_toolchains": ["uARM", "ARM", "GCC_ARM", "IAR"],
        "inherits": ["Target"],
        "device_has": ["ANALOGIN", "I2C", "INTERRUPTIN", "PORTIN", "PORTINOUT", "PORTOUT", "PWMOUT", "RTC", "SERIAL", "SPI", "SPISLAVE", "STDIO_MESSAGES"],
        "release_versions": ["2", "5"]
    },
    "WIZWIKI_W7500ECO": {
        "inherits": ["Target"],
        "core": "Cortex-M0",
        "extra_labels": ["WIZNET", "W7500x", "WIZwiki_W7500ECO"],
        "supported_toolchains": ["uARM", "ARM", "GCC_ARM", "IAR"],
        "device_has": ["ANALOGIN", "I2C", "INTERRUPTIN", "PORTIN", "PORTINOUT", "PORTOUT", "PWMOUT", "RTC", "SERIAL", "SPI", "SPISLAVE", "STDIO_MESSAGES"],
        "release_versions": ["2", "5"]
    },
`

The microcontrollers are w7500 : WIZWIKI_W7500ECO, WIZWIKI_W7500
and w7500p: WIZWIKI_W7500P
We need gcc-Support for using with PlatformIO

theotherjimmy, 2017-06-01:
@sstaub Those devices don't have a `device_name` key in their target data. When I commented last, having that field filled in was required to export for uvision5. 

At the moment, the devices should export correctly _offline_ for eclipse_armc5 , eclipse_gcc_arm , eclipse_iar , embitz , gnuarmeclipse , make_armc5 , make_gcc_arm , make_iar , qtcreator , uvision , vscode_armc5 , vscode_gcc_arm and vscode_iar

sstaub, 2017-06-01:
Thank you, we will check it on next releases of PlatformIO

theotherjimmy, 2018-06-14:
@sstaub Is this resolved then?

sstaub, 2018-06-14:
Yes it works

closed 2018-06-14 by sstaub