ARMmbed/mbed-os #3363: Feature request: DFU

shayo, 2016-12-03:
- Type: Enhancement

I would very much like to transition to mbed OS5. However, it seems like DFU is not yet supported, which is a major blocker.
I would like to ask to include DFU support in OS5 along with:
1. single bank bootloader to allow large applications to be loaded
2. bootloader with watchdog support.

In general, it would be also nice to have improved method for selecting which bootloader to use in the online compiler IDE.

**Target**
NRF51DK

**Toolchain:**
GCC_ARM

**mbed-cli version:** 
0.9.10

BlackstoneEngineering, 2016-12-04:
Clarification, you mean Over The Air DFU for BLE, correct?

shayo, 2016-12-04:
Correct

pan-, 2016-12-05:
@nvlsianpu Could you comment on this issue ?

shayo, 2016-12-05:
Here is what I got from Andrea Corrado over email (Nov 21, 11:40 GMT):

> "The NRF51 DK does **not** support DFU on the latest revisions of mbed; at the moment. Old examples should still work.
> Nordic engineers are working on bringing back DFU and I’m helping them.

My attempts to update NRF51DK with nrftoolbox have all failed when I upload OS5 type applications. They work well with applications compiled with mbed-classic.

nvlsianpu, 2016-12-06:
Work on Bootloader for nRF5x @s13.x 2.x.x are in progress already. I must consider having either single bank and dual bank solutions apart from dual bank. Bootloader almost idenitical to those available in Nordic nRF5 SDK 11 will be provided in first step (http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk51.v10.0.0%2Fexamples_ble_dfu.html&cp=4_0_2_4_3_1). Unfortunately there is no watchdog support incorporated in Nordic DFU examples. @shayo It will be helpful for me if you provide some more description on the requested watchdog feature.

shayo, 2016-12-06:
The watchdog feature is really important. Many apps use watchdog to make sure their app doesn't get stuck. Once this feature is activated, it cannot be turned off. Since DFU takes quite a bit of time, the dog needs to be kicked once in a while. Otherwise, the watchdog will kill the DFU process.
There has been several threads on Nordic forum on how to implement this. Please refer to:
https://devzone.nordicsemi.com/question/56888/wdt-in-bootloader/
https://devzone.nordicsemi.com/question/818/watchdog-timer/

Is there any timetable for releasing the new bootloader?
Many thanks for considering those features!

shayo, 2016-12-06:
In what ways are the bootloaders that you are working different than the default example provided in the SDK? What needs to be changed in order to make it work with OS5?

nvlsianpu, 2016-12-06:
>In what ways are the bootloaders that you are working different than the default example provided in the SDK?

LED and button handling are removed. Rest of the code without changes.

FYI https://github.com/ARMmbed/mbed-os/pull/3376

nvlsianpu, 2016-12-07:
I build single bank bootloader with WDT handling  using this code https://github.com/nvlsianpu/nrf5_bootloader. It is a POC - contributions of improvements are highly welcomed (unfortunately I can't spend much time on this). This bootloader is incorporated in NRF51_DK_BOOT_WDT which was added here https://github.com/nvlsianpu/mbed/tree/nrf5_dfu_s13x_wdt.

shayo, 2016-12-07:
Dear @nvlsianpu, 
Thank you for the effort. I will try to compile and test those today.

shayo, 2016-12-07:
@nvlsianpu , what would be needed to make NRF51_DK_BOOT_WDT work also for NRF52?

shayo, 2016-12-07:
@nvlsianpu , I am getting the following error message when trying to compile (even the regular NRF_51DK_BOOT):

...
Compile [100.0%]: test_env.cpp
Link:Proj_OS5
Elf2Bin: Proj_OS5
**[ERROR] Data overlapped at address 0x3C000**
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /Users/shayo/mbed/Proj_OS5/mbed-os/tools/make.py -t GCC_ARM -m NRF51_DK_BOOT --source . --build ./BUILD/NRF51_DK_BOOT/GCC_ARM" in "/Users/shayo/mbed/ProjR07_OS5"

Using the latest mbed master branch I did not get this data overlap error.
What does it indicate? that the app is too big and collides with the softdevice?

nvlsianpu, 2016-12-07:
Did you try to run a clean build?
>mbed compile -m NRF51_DK -t GCC_ARM --vv -c

Data overlap could been cussed by your application's size (nrf51 bootloader code start form 0x3c000 - so it deduct 16 KB form flash available for an application). You could use additionally option
 >--profile small

in a mbed  build command (it use settings from here https://github.com/ARMmbed/mbed-os/blob/master/tools/profiles/small.json) in order to apply min-size settings for your build.

shayo, 2016-12-07:
OK. I will try that. I also tried merging some of your changes to build 130 of the master branch, but i keep getting this error after the compilation:
...
Compile [100.0%]: test_env.cpp
Link: Proj
Elf2Bin: Proj
[ERROR] expected a string or other character buffer object

Is there a way to compile with verbose to see exactly where and why this crash occurs?

shayo, 2016-12-07:
Ah. I see. "-v" does the job. OK. let me look into that error and I'll report back.

shayo, 2016-12-07:
OK. Thank you for those tips @nvlsianpu !
The crash about a missing string was because I didn't merge your changes to the targets.py.
Also, adding the --profile small solved the overlap problem. Seems like the project compiles. I will go ahead and test it on the real device later tonight.
Can you provide more information on what the --profile small is doing? Are those features documented somewhere?

nvlsianpu, 2016-12-07:
It apply some setting for compiler, linker, asm etc. see https://github.com/ARMmbed/mbed-os/blob/master/tools/profiles/small.json#L3-L14.

be aware of https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v11.0.0%2Fbledfu_bootloader_introduction.html Probably you will need to creating a zip with image and init packet.

shayo, 2017-02-04:
@nvlsianpu - would it be possible to release a WDT single bank bootloader for S130 v1.0.0? (i.e., one that supports v0.4 bootloader transfer - no init file).
I've been trying to compile the original project (https://github.com/ARMmbed/nrf5x-dfu-bootloader), but I'm getting many errors due to wrong libraries used (I am not sure which ones were originally used to compile that project). 
I was hoping I could switch to OS5 but now it looks like the target size will be too big for nRF51822 (the only way I can fit the firmware on the device is if I use the legacy code).
I could really use your help in reviving that code and compiling it with WDT support.
Many thanks,

Shay

nvlsianpu, 2017-02-06:
Nordic doesn't support s130 v1.0.0 any more. Instead of compiling nrf5-dfu-bootloader you could  compile single_bank_wdt_ble_s130 form SDK 10.0.0 and use this as bootloader code (nrf5-dfu-bootloader - it is just the same as in SDK make/gcc project setup)

james-hatch, 2017-03-11:
What is the status of DFU OTA for NRF52? Is there a time estimate of when this might be released?

nvlsianpu, 2017-03-13:
For current NRF52 target DFU is on [feature branch of mbed-os](https://github.com/ARMmbed/mbed-os/tree/feature-nrf5_dfu_s13x_v2/targets/TARGET_NORDIC) - so you can evaluate this. Probably ARM is going to work a little on BLE-gatt-service interface for dfu.

> What is the status of DFU OTA for NRF52? Is there a time estimate of when this might be released?

@pan- Do you know when this might be released?

shayo, 2017-05-18:
Any news on this?
I just downloaded the latest os5 and it seems like there is still no NRF52_DK_BOOT or NRF52_DK_OTA

seppestas, 2017-06-08:
We have been using the `feature-nrf5_dfu_s13x_v2` feature branch successfully. The bootloader seems to work, but it changes the [Bluetooth MAC address of the NRF52 when going into DFU mode](https://devzone.nordicsemi.com/question/67844/why-does-nordic-increase-the-ble-mac-address-by-one-in-dfu-bootloader/) since bonding does not seem to work on the feature branch.

I'm implementing OTA DFU on top of mbed OS 5.4.7, but I'm still doubting between sticking to the "Legacy" DFU designed to work with the NRF SDK v11, or switching to the more modern "Secure DFU" bootloader on top of the NRF SDK v13.

Both v11 and v13 of the NRF SDK are currently inside of mbed, but as far as I can tell v13 is only used for the `MCU_NRF52840` target, while the `MCU_NRF52` target still uses v11.

Are there any important reasons to stick to v11 on the NRF52832? @pan- is there already some kind of roadmap I could try to conform to? A challenge might be the extra pre-compilation step required to support the proto files, but I think the mbed toolchain should be capable of handeling this.

@nvlsianpu could you give me some more information about how the `sdk_update` tool and maybe tell me why `ble\\ble_services\\ble_dfu` [seems to be absent](https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_SDK13/ble) even though [it is listed in `update_desc.json`](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5/porting_tools/update_desc.json#L108)? We have a contact at Nordic, but he does not seem involved in the mbed story.

nvlsianpu, 2017-06-08:
sdk_update is only a __helper__ script. NRF52840 support had internally started form sdk11 base which was updated to 13xxalpha. As I remember after I had used (for update) it I made changes by hand too - e.g. i remove unused ble_dfu modules. I expect that update_desc.json must be adapted for each update.

Other thinks is that Secuer DFU is much different from previous one - So it require lot of modification in mbed and BLE DFU service.

Don't hesitate to ask Nordic client support contacts - we always need feedback form ours customers, Sometime is much better to poke them than developers - especially if a feature is needed.

seppestas, 2017-06-08:
@nvlsianpu From what I understand secure DFU mostly impacts the bootloader and the deployment packages. However, it looks like the buttonless DFU example in v13 and v12 of the Nordic SDK does not support bonded devices since there is no way to communicate the LTK with the bootloader. If I understand correctly, this means that using "buttonless" OTA DFU using an iOS programmer will not work. Can you tell me if this is a limitation of the bootloader or if there is actually a`dfu_ble_svc_set_peer_data` like function for the v13 SDK / bootloader that is not exposed?

Since we want to use buttonless DFU I will first be looking into getting the bootloader of the v11 SDK to work based on the example in the SDK, on top of the changes introduced to the BLE Security Manager.

seppestas, 2017-06-08:
Update: it looks like I can merge the changes you made on `feature-nrf5_dfu_s13x_v2` onto `mbed-os-5.4.7`.  @pan-, why hasn't the `feature-nrf5_dfu_s13x_v2` branch been merged yet? The [security extensions for the BLE_API](https://developer.mbed.org/forum/team-63-Bluetooth-Low-Energy-community/topic/5661/) do no seem to be in the feature branch and they are very usefull in combination with OTA DFU.

nvlsianpu, 2017-06-08:
Since you ask detailed questions regards DFU itself - I cant assist - please ask us via devzone.

nvlsianpu, 2017-06-08:
BTW https://github.com/nvlsianpu/nrf5_bootloader - this is how bootloader image was produced for mbed.

seppestas, 2017-06-09:
Looks like storing the peer data does not work due to the introduction of the "SDK nRF5 Peer manager" in e67117b8ea. This causes the `dfu_transport_ble` to increment the mac address, causing "transparent" buttonless dfu not to work on mbed-os.

As far as I can tell, [the bootloader still uses the DM for accessing the peer data](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_SDK11/ble/device_manager/device_manager_peripheral.c#L2441), not the PM. This causes the bootloader to fall back on the "increment mac address" method (see dfu_transsport_ble.c:1032 in SDK v11).

I guess the best solution would be to have the bootloader use the PM to get the keys, but since this requires re-compilation of the bootloader I'm first going to test if falling back on the DM works. I think it would be interesting to have the mbed toolchain compile and link in the bootloader instead of relying on the precompiled bootloader. @nvlsianpu any thoughts on this?

anweiss, 2017-07-17:
Any idea when the `feature-nrf5_dfu_s13x_v2` branch will be merged and what version of mbed this will make it in?

anweiss, 2017-07-22:
@pan- ? ^

jrobeson, 2017-08-16:
yes it'd be nice to know when this might appear.

janjongboom, 2017-11-23:
@pan- @sg ping on this...

jrobeson, 2018-04-23:
so what's left to do to get DFU for the nrf targets?

janjongboom, 2018-04-24:
@marcuschangarm Have you been looking at this?

melvinvdb, 2018-05-09:
Am I correct that mbed 5.8 still doesn't support DFU on the nrf51? 
I'm confused because NRF51_DK uses softdevice v2 without bootloader support and NRF51_DK_BOOT uses v1 with bootloader but it seems to be broken. :/

jrobeson, 2018-06-14:
I'm still confused as to what's missing. mbed-os now does this:

```sh
Using regions bootloader, application in this build.
  Region bootloader: size 0x23000, offset 0x0
  Region application: size 0x5d000, offset 0x23000
```

which certainly brings things closer. Is it it fixes to the DFU service itself, or something else?

janjongboom, 2018-06-19:
@marcuschangarm ^

marcuschangarm, 2018-06-19:
@jrobeson That is the SoftDevice being merged.

jrobeson, 2018-06-20:
@marcuschang: that's not really answering the question though :)

jrobeson, 2018-07-21:
oops bad completion in my previous message a month ago.

@marcuschangarm: that's not really answering the question.  What's missing for DFU support in mbed-os master?

kaidokert, 2018-08-12:
This should document a working setup for getting DFU: 

https://os.mbed.com/users/kord123/notebook/fota-mbed-app-to-nrf52_dk-with-nordic-dfu-boot-loa/

jrobeson, 2018-08-12:
I was under teh impression that the current DFUService needed to be updated to use the new peermanager apis or something like that.

jrobeson, 2018-10-25:
can someone respond to my previous comment re: the DFUService?

closed 2019-10-31 by ciarmcom