ARMmbed/mbed-os #3364: why is "No lock needed in the constructor"?

odinthenerd, 2016-12-03:
here the constructor does not lock https://github.com/ARMmbed/mbed-os/blob/master/drivers/DigitalInOut.h#L41 however there are many chips where the actions in the constructor would be race conditions if interrupted by a context switch or ISR. What exactly is so special about the constructor that it does not need to lock?

```
//one thread
DigitalInOut pin1(/*some pin*/);

//other thread
DigitalInOut pin2(/*other pin*/); //if context switch somes at the right time this is race condition

//maybe in an ISR
pin1.input(); //if this interrupts the constructor of pin2 then race condition
```

0xc0170, 2016-12-05:
cc @c1728p9

c1728p9, 2016-12-05:
The initial design required that the HAL implementation initialized GPIO (or another peripheral) in such a way that it didn't effect other GPIO pins. Most MCUs have write only registers to set the state in addition to a dedicated register for each pin to configure it.

Currently there are not HAL tests to verify this behavior. Since the overhead of adding locking to initialization is small, it probably doesn't hurt to add.

odinthenerd, 2016-12-05:
what about here? https://github.com/ARMmbed/mbed-os/blob/master/drivers/CAN.cpp#L27

or here? https://github.com/ARMmbed/mbed-os/blob/master/drivers/SPI.cpp#L27

It seems to me that either you dictate that everything from the vendors are reentrant, and enforce it, as opposed to the past, or you lock. Arbitrarily locking on most stuff but not on constructors seems unlogical as most of the bitwise false sharing stuff is during init.

odinthenerd, 2016-12-05:
Actually looking at those again the whole locking concept is wrong in some cases. https://github.com/ARMmbed/mbed-os/blob/master/drivers/SerialBase.cpp#L128 here you are taking a local lock in order to stop corruption from happening lets say here https://github.com/ARMmbed/mbed-os/blob/0ace7955b0949fce0448269bc11d64eca05da24d/targets/TARGET_NXP/TARGET_LPC15XX/serial_api.c#L102

The local lock does nothing to prevent bitwise false sharing between different modules which have their own local lock. You need to be preventing context switching/ISRs here. That in turn makes reaction time terrible but that's what you think of concurrency first and not years after the fact ;)

c1728p9, 2016-12-05:
It would be great to have the speed increases mentioned in your post. I don't know how feasible this is in practice, as all the vendor provided C HALs would need to be re-written. If there was a straightforward way to incrementally upgrade the C HAL APIs to the faster version you propose this might be possible. Would you be able to put together a draft which makes this change for at least one target?

As for the locks, the mbed library provides thread safety, but the user must be still reasonable about the use of components. For example, two Serial instances on the same set of pins but independent uses is a logic error and will cause problems regardless of if they share a common lock. Components which are reasonable expected to be have multiple instances such as SPI and IIC use a global lock to gracefully handle multiple instances being used from different threads. Because of the different use cases different locking paradigms are intentionally used. I would not say either of these concepts are wrong. There very well could be bugs which I would be happy to fix.

odinthenerd, 2016-12-05:
Yeah ok fair enough, its kind of a stretch. However we have hooked a capture input to a uart rx pin in a design in order to adjust the baud rate. We also connected a gpio interrupt to a CAN rx in order to wake up from sleep. It is a niche use case though and not even possible on all chips.

what about this more probable scenario https://github.com/ARMmbed/mbed-os/blob/a8ebfaa0584b183cba3bed7fa540b424d6b7ade8/targets/TARGET_STM/TARGET_STM32F0/i2c_api.c#L294 interrupts https://github.com/ARMmbed/mbed-os/blob/6eb33e5f3cae9d22ed5a7ec0dcf56d01acdbaae2/targets/TARGET_STM/TARGET_STM32L0/serial_api.c#L106

these are two separate modules with two separate locks which share a SFR (in this case RCC->APB2RSTR) which they do not just touch at init. So just protecting init is not always enough. Thats what I mean when I say this strategy is broken. 

[better api]
I will have a look, I would like to make the nordic chips thread safe and much faster. I will look into what changes need to be made to the api in order to make that work and maybe we can work something out. By the way what compilers does it have to support? (in other words how much TMP can I use)

theotherjimmy, 2016-12-06:
@odinthenerd You have to support arm-none-eabi-gcc, ARM Compiler 5.06u2, and IAR 7.60.2

theotherjimmy, 2016-12-06:
@odinthenerd  https://github.com/ARMmbed/mbed-os/blob/master/platform/toolchain.h may help

odinthenerd, 2016-12-06:
IAR 7.60.2 says it support C++03, if it really does (including all the template stuff) one could actually do a lot of compile time decisions based on the traits of the vendor's HAL. This would be a non breaking change as far as vendor code goes but it would give them a path to improve. I'll look into making a demo, maybe over Christmas.

theotherjimmy, 2016-12-06:
@odinthenerd I look forward to seeing what you put together.

closed 2017-02-04 by sg-