ARMmbed/mbed-os #3370: Atmel SAMD target blinky won't compile

malachib, 2016-12-06:
`mbed compile` fails looking for "mbed_rtx.h" file which apparently isn't present for TARGET_Atmel

0xc0170, 2016-12-07:
Atmel devices were not supported by RTOS previously. Therefore there's no header file.

We could provide the error message that RTX is not supported. but that header file with the error message would need to be compiled only if mbed-cli (mbed OS 5) is being used..

malachib, 2016-12-07:
OK, so Thread::wait makes the example code https://github.com/ARMmbed/mbed-os-example-blinky RTOS specific it seems.  Should we expect to be able to compile for SAMD if we avoid RTOS/RTX references?

MINAR was cool, but not supported now; I have an mbed_events based blinky sample, but compiling it produces the same RTX related error:

```
#include <mbed.h>
#include <mbed_events.h>

#define MAL_LED PA17

static void blinky(void) {
    static DigitalOut led(MAL_LED);
    led = !led;
    printf("LED = %d \r\n",led.read());
}

int main()
{
  EventQueue queue;

  queue.call_in(2000, printf, "called in 2 seconds!");
  queue.call_every(1000, blinky);
}
```

Gives us:

```
Compile [ 74.4%]: RTX_Conf_CM.c
[ERROR] ./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_Conf_CM.c:43:22: fatal error: mbed_rtx.h: No such file or directory
```

0xc0170, 2016-12-08:
@malachib using mbed OS 5, RTOS in included by default, therefore Atmel devices are not yet supported.

malachib, 2016-12-08:
OK, I understand.  Dang!  Well, hopefully soon.  No chance of an mbed OS 5 build with RTOS disabled?  Not that I even think that's a great idea, just curious...

bridadan, 2016-12-08:
@malachib This would be mbed OS 2, which is still being release on the developer site. Here's the link to the library's page: https://developer.mbed.org/users/mbed_official/code/mbed/

malachib, 2016-12-09:
Now that is a response I didn't expect. lol thank you though
Your answer makes me want to ask more questions!
- mbed OS 3 looked promising and supported the target.  Evidently there was no RTX/RTOS support there either, yet had the newer API.  I would have expected that same level of functionality to roll forward to 5.x.  What happened?
- how future proofed is one when investing in mbed OS 2?
I know I can be annoying with all these questions, I have just been having a lot of trouble getting my footing in this environment, so thanks for your patience.

0xc0170, 2016-12-09:
> mbed OS 3 looked promising and supported the target. Evidently there was no RTX/RTOS support there either, yet had the newer API. I would have expected that same level of functionality to roll forward to 5.x. What happened?

there were some features ported . What are you missing?

> how future proofed is one when investing in mbed OS 2?

cc @sg-

bridadan, 2016-12-09:
@malachib Nothing wrong with questions! I'll try to answer your first one at least.

> mbed OS 3 looked promising and supported the target. Evidently there was no RTX/RTOS support there either, yet had the newer API. I would have expected that same level of functionality to roll forward to 5.x. What happened?

I'm afraid I can't comment on that directly since I wasn't really involved with mbed OS 3. What I can say is the API from mbed OS 3 (MINAR) was dropped in favor of the RTOS for mbed OS 5.

Though it's not exactly the same, you may be interested in the events API that was introduced in mbed OS 5.2. It offers a similar way of writing applications. You can read more about the events API here: https://docs.mbed.com/docs/mbed-os-api-reference/en/5.2/APIs/tasks/events/

malachib, 2016-12-09:
@bridadan I've tried out the events API before, and it fully suits my needs.  However, the trouble is that RTOS/RTX is included by default - so even if I use events API, I am unable to compile.

@0xc0170 the "same level of functionality" I may have misled you, I was only musing on Atmel SAMD targets going away because new functionality was added where that didn't seem to be the case for mbed 3.0.  I have a resistance to going to 2.0 API just 'cause I feel like it's going "backwards" - with all due respect to the hard work still going on on 2.0 API

janjongboom, 2016-12-19:
@malachib Here's some pointers how to remove RTX: https://developer.mbed.org/blog/entry/Reducing-memory-usage-by-tuning-RTOS-con/

malachib, 2016-12-19:
@janjongboom So it's possible after all!  Nice!

malachib, 2017-05-08:
Blinkies are working very well, esp since 5.4

A few thoughts & questions:

* Should we close this issue out, or does disabling RTOS not count as an "official" solution?
* I'd like to ask what the impediments are for bringing RTOS to SAMD devices
* My blinkies work great now with SAMD, but semihosting is basically nonexistant.  Linux doesn't even see a USB device when the flashed mbed firmware is running.  However, that might be because I am fiddling around and using BOSSAC to flash with.  What would my first step be to getting USB serial logging back online?

PTS93, 2017-08-21:
SAMD21 support would indeed be really nice. After all there are a LOT of dev boards out there that use it and its a very price competitive chip in low quantities, very interesting for custom developments.

malachib, 2018-10-02:
Might I inquire where this JIRA is so I can monitor any progress?

PTS93, 2018-10-03:
There is also the SAMD51 now which is a Cortex M4 and nearly pin compatible and software wise the same except for the newer peripherals.

closed 2019-10-31 by ciarmcom