ARMmbed/mbed-os #3371: Socket events lack parameters

bogdanm, 2016-12-06:
It is currently possible to send events for various socket operations (like send and receive), but the event doesn't carry around any information about the socket that generated the event, or what kind of event was generated: 

https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/Socket.h#L16

This makes it impossible to handle the state of an application that receives events for more than one socket, since the application can't know which socket generated the event. Moreover, the comment for the `attach` function above states:

> The callback may also be called spuriously without reason.

which makes keeping a consistent state in the application even harder.

@marcuschangarm

bogdanm, 2016-12-06:
Let's simplify things a bit, this was written initially after looking at some C (not C++) code that used a single handler for all its sockets, which can be fixed by attaching a separate handler to each socket. But the other issue remains: it would be nice for the event to have a parameter that specified which operation it was sent for, like `EVENT_SEND` or `EVENT_RECEIVE` (especially important because of the "spurious" part above).

geky, 2016-12-06:
I would hesitate to call the attach callback for the socket API an asynchronous API. It can be used asynchronously, but it was originally added as an alternative to busy waiting in applications that use an FSM to drive a network protocol (ie mbed-client). The decision to allow spurious wakeups was to make porting new devices easier (ie esp8266, mx, c027, etc), and when the callback is triggered is rather sporadic. For example, the esp8266 triggers the callback whenever data is ready on the serial line. This occurs quite often, but still allows the device to sleep when there is no network transactions.

The intended use case for the attach callback is to trigger a signal or event loop that polls a socket’s recv or send. From this polling (which will never block if the socket is set to non-blocking) the application can be more information about the state of the socket.

---

All of that being said, it would be nice to have an asynchronous socket API with precise behavior. I was looking into extending the socket’s asynchronous API in a manner similar to CAN/Serial, but it isn’t a high priority (it’s here https://github.com/geky/mbed/tree/async-api, though I don’t remember if it compiles, and I wouldn’t be opposed to a complete rewrite).

closed 2017-10-27 by ChiefBureaucraticOfficer