ARMmbed/mbed-os #3375: mbed-os-5.3.0-rc2: export make_iar for Odin board fails

bulislaw, 2016-12-06:
### Description
- Type: Bug
- Priority: Major

## Bug

**Target**
UBLOS_EVK_ODIN_W2
**Toolchain:**
IAR

**Toolchain version:**
 IAR ANSI C/C++ Compiler V7.70.1.11437/W32 for ARM

**meed-os sha:**
mbed-os-5.3.0-rc2

**Expected behavior**

Compile and link properly

**Actual behavior**
The compilation succeeds, but linkage fails:
```
$ make
link: mbed-os-example-wifi.elf

   IAR ELF Linker V7.70.1.11437/W32 for ARM
   Copyright 2007-2016 IAR Systems AB.
Error[Li005]: no definition for "gpio_write" [referenced from c:\Users\barsza01
          \mbed\mbed-os-example-wifi\BUILD\mbed-os\platform\mbed_board.o]

   54 960 bytes of readonly  code memory
    1 220 bytes of readonly  data memory
  104 040 bytes of readwrite data memory

Errors: 1
Warnings: none

Link time:   0.56 (CPU)   1.36 (elapsed)
make[1]: *** [mbed-os-example-wifi.elf] Error 2
make: *** [all] Error 2
```

@theotherjimmy @sarahmarshy

sarahmarshy, 2016-12-06:
Are you able to compile this example with `mbed compile` with IAR?

bulislaw, 2016-12-07:
Yeah, `mbed compile` with IAR works fine.

theotherjimmy, 2016-12-20:
Whatever the cause was, It's fixed in 5.3.0-rc3

bridadan, 2016-12-20:
@bulislaw can you confirm? Otherwise we'll close this issue.

closed 2017-02-04 by sg-