ARMmbed/mbed-os #3383: Use the ‘u’ and ‘d’ keys to make LED1 brighter or dimmer (from handbook) broken on K22F

AnotherButler, 2016-12-07:
### Handbook example to use the 'u' and 'd' keys to make LED1 brighter or dimmer doesn't work on K22F. See example on page: https://docs.mbed.com/docs/mbed-os-handbook/en/5.2/getting_started/serial_communication/
- Type: Bug
---------------------------------------------------------------
## Bug

**Target**
K22F

**Expected behavior**
Pressing 'u' and 'd' keys makes LED1 brighter and dimmer

**Actual behavior**
LED1 doesn't get dimmer, but it should.

**Steps to reproduce**
Go to handbook page: https://docs.mbed.com/docs/mbed-os-handbook/en/5.2/getting_started/serial_communication/. Do example toward bottom of page, Use the ‘u’ and ‘d’ keys to make LED1 brighter or dimmer, on K22F. Other boards seem to work.

0xc0170, 2016-12-08:
cc @mmahadevan108

bridadan, 2016-12-08:
Many of the boards (K64F as well) don't have the LED1 on PWM pin, so we should probably remove this example from the handbook.

closed 2016-12-21 by AnotherButler