ARMmbed/mbed-os #3384: mbed compile's --cflags option

Nodraak, 2016-12-07:
### Description
- Type: Question
- Related issue: #2431 and #2802
- Priority: Minor

---------------------------------------------------------------
## Question

(mbed-cli version 0.9.1 / meed-os sha ea3526d)

**How to** pass flags to the C and CPP compilers (like `-E` or `-O0` or `-mfloat-abi=soft`). I saw the  mbed compile's `--cflags` option, but it is not taken into account (I can see that thanks to the mbed compile's `-v` flags that print the actual executed command).

From the issues of this repo I saw that it might have been replaced by *profile files* but I have no clue about how to write this damn json (where, what name, ...).

Anyway, if the cli accept the `--cflags` option, I would expect it to be took into account or at least show a warning if it was to ignore them silently.

Thanks for your help !

theotherjimmy, 2016-12-07:
Hey @Nodraak! 

>From the issues of this repo I saw that it might have been replaced by profile files but I have no clue about how to write this damn json (where, what name, ...).

The documentation you are looking for is here: https://github.com/ARMmbed/mbed-os/blob/master/docs/Toolchain_Profiles.md . TL;DR: pass the filename to `mbed compile`'s `--profile` option. e.g. `mbed compile -t gcc_arm -m k64f --profile my_profile.json` would build you project with arm-none-eabi-gcc, for the frdm-k64f, using the profile found in my_profile.json.

There are three examples in the codebase: https://github.com/ARMmbed/mbed-os/tree/master/tools/profiles

You could copy one of these files and modify it to suit your needs.
> Anyway, if the cli accept the `--cflags` option, I would expect it to be took into account or at least show a warning if it was to ignore them silently. 

Agreed. That's why I created https://github.com/ARMmbed/mbed-os/pull/3385 .

theotherjimmy, 2016-12-16:
#3385 merged. Closing.

closed 2016-12-16 by theotherjimmy