ARMmbed/mbed-os #3386: error: 'struct NRF_TWI_Type' has no member named 'POWER'

shayo, 2016-12-07:
- Type: Bug / Question
- Priority: Minor

## Bug

**Target**
NRF52_DL

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
0.9.10

mbed-os:
master branch (version 130)

I have a program with my own implementation of I2C two wire communication.
It compiles for target NRF51DK, but for NRF52DK I get the following error:

./Wire/wire.cpp:256:18: error: 'struct NRF_TWI_Type' has no member named 'POWER'

What happened to the power field? has it been dropped? changed?

nvlsianpu, 2016-12-08:
Hi.
POWER field had been removed in nrf52832. NRF52 has automated power/clock control unit, so POWER reg. is no need for going to LP mode. See 49.7 Low power in NRF52832 OPS.

0xc0170, 2016-12-08:
@shayo If the above comment answers your question, please close the issue. Thanks

closed 2016-12-08 by shayo

curtpw, 2017-05-09:
So what did you do to solve the problem?

shayo, 2017-05-09:
I think I just removed those lines. I haven't tested things throughly. Let us know if that fixed the issue.