ARMmbed/mbed-os #3395: Enable broadcast on UDPSocket on LPC1768

WillyKaze, 2016-12-08:
## Description
- Type: Question
- Priority: Blocker

## Question

**How to enable broadcast option on a UDPSocket?**

I'm struggling to find an answer to this question.

I managed to find an old solution `sock.set_broadcasting();` in some example but this method is not defined anymore in mbed-os 5.2. This method is still defined in `features/unsupported/net/lwip/Socket/UDPSocket.cpp` thought.
I tried to have a look at lwip feature but seems to rely on nsapi a lot and nsapi doesn't define the `SO_BROADCAST` socket option.

Did anyone faced a similar problem ? Is there a workaround ?

Best regards,
Ben

0xc0170, 2016-12-08:
cc @geky @kjbracey-arm

sg-, 2017-02-04:
bump @geky @kjbracey-arm

closed 2017-10-27 by ChiefBureaucraticOfficer