ARMmbed/mbed-os #3396: Thread not working at all???

jsimunic, 2016-12-08:
### Description
- Type: Bug
---------------------------------------------------------------
## Bug

**Target**
C027

**Toolchain:**
mbed online compiler

-----------------------------------------------------------------

## Question

**How to start running a thread with Thread::start() ?**

The example in the docs for the thread (from the Thread.h) is not working at all.
Compiled on the mbed online compiler with the newest version of mbed-os imported directly from github.

`#include "mbed.h"`
`#include "rtos.h"`

`Thread thread;`
`DigitalOut led1(LED1);`
`volatile bool running = true;`

`// Blink function toggles the led in a long running loop`
`void blink(DigitalOut *led) {`
`    while (running) {`
`        *led = !*led;`
`        Thread::wait(1000);`
`    }`
`}`
`// Spawns a thread to run blink for 5 seconds`
`int main() {`
`     thread.start(led1, blink);`
`     Thread::wait(5000);`
`     running = false;`
`     thread.join();`
`}`

**Compiler error number:** 304
**Compiler error description:** No instance of overloaded function "rtos::Thread::start" matches the argument list "thread.start(led1, blink);"

0xc0170, 2016-12-08:
I can confirm it does not compile. I suspect there were changes to Thread class, that have not been reflected to the example provided in the header file. It should be using start with a callback object as a argument,

cc @geky

0xc0170, 2016-12-08:
Seems like this is related - https://github.com/ARMmbed/mbed-os/issues/3203. We should fix both

sarahmarshy, 2016-12-19:
>`thread.start(led1, blink);`

Instead of the above, you'll need to use references like this:

>`thread.start(&blink, &led1);`

Though, the above is deprecated in favor of using callbacks like this:

>`thread.start(callback(&blink, &led1));`

Thread::start should always be passed a callback. The order of parameters to the callback is either:
`callback([object to invoke member function], [member function])`
OR 
`callback([static function], [pointer argument to function])`

Callback class templating [here](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.2/platform/Callback.h#L125) .

jsimunic, 2017-01-12:
Thanks for resolving this issue quickly, tested it on the C027 and the KL25Z, it works on the KL25Z but not on the C027 (the LED doesn't blink at all). Tested on the C027 with three test cases:
First case used the onboard LED, second case with an external LED connected to P0_1, third case with an external LED connected to P1_20.
@0xc0170 @sarahmarshy

jsimunic, 2017-01-12:
Forgot to say that including rtos.h is not necessary since it is already a part of mbed-os.

0xc0170, 2017-01-12:
@jsimunic The test example to reproduce the problem is the same as above?

cc @mazgch

jsimunic, 2017-01-12:
@0xc0170 Well I used the example above with the changes that @sarahmarshy  did in #3469 in the mbed online compiler with mbed-os (tag:mbed_lib_rev133).
Here it is: 
```
#include "mbed.h"
#include "rtos.h"

Thread thread;
DigitalOut led1(LED1);
volatile bool running = true;

// Blink function toggles the led in a long running loop
void blink(DigitalOut *led) {
    while (running) {
    *led = !*led;
    wait(1);
    }
}
// Spawns a thread to run blink for 5 seconds
int main() {
    thread.start(callback(blink, &led1));
    wait(5);
    running = false;
    thread.join();
}
```

In all of the test cases I changed only line 5 where we instantiate the DigitalOut object:
First case:
`DigitalOut led1(LED);`
Second case:
`DigitalOut led1(P0_1);`
Third case:
`DigitalOut led1(P1_20);`

sg-, 2017-02-03:
@jsimunic Did this get resolved or clairified?

jsimunic, 2017-02-04:
@sg- Still nothing, didn't try to contact guys from u-blox, haven't tried out the test cases on a LPC1768 (the MCUs are the same on C027 and a LPC1768 only different package, right?) but I will in the following weeks when I gain access to a LPC1768.

closed 2017-10-27 by ChiefBureaucraticOfficer