ARMmbed/mbed-os #3398: nrf52 - UARTE not available

0xc0170, 2016-12-08:
### Description
- Type: Question
- Priority: Major

## Question

nrf52 has 2 UART peripherals (UART and UARTE). mbed provides HAL implementation for UART only, therefore in the peripheral map theres only one UART. This UART however is used as stdio. Thus a user does not have any free uart available. For instance, if there's serial wifi shield, it can't be used out of the box.

I checked the reference manual, looks to me there's UARTE available. That could be used by an application.

Are there any limitations ?

@nvlsianpu

pan-, 2016-12-08:
@0xc0170 Actually, I've checked and the UARTE peripheral is a super set of the UART peripheral (deprecated) they both share the same peripheral instance. I don't think it will be possible to use both.

0xc0170, 2016-12-08:
> @0xc0170 Actually, I've checked and the UARTE peripheral is a super set of the UART peripheral (deprecated) they both share the same peripheral instance. I don't think it will be possible to use both.

I haven't noticed that one, that would mean none UART peripherals available for users 😞

nvlsianpu, 2016-12-08:
It's true - It is impossible to use UART & UARTE - it use partialy the same hardware and registers.

0xc0170, 2016-12-09:
Thank you for your answers ! It's pity it does not have more than 1 uart peripheral :sob: 

Closing this as resolved.

closed 2016-12-09 by 0xc0170

nvlsianpu, 2016-12-09:
BTW - if you use j-link firmware you can use RTT instead of uart. We have this solution in nRF5 SDK.