ARMmbed/mbed-os #3400: [Beetle] BLE scan doesn't work

pan-, 2016-12-08:
### Description
- Type: Bug 
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
Beetle board 

**Toolchain:**
GCC_ARM|ARM|IAR

**Steps to reproduce**

Run the [EddystoneObserver](https://github.com/ARMmbed/mbed-os-example-ble/tree/master/BLE_EddystoneObserver) example while another board run the [EddystoneService](https://github.com/ARMmbed/mbed-os-example-ble/tree/master/BLE_EddystoneService) example. 
The observer should print detected Eddystone beacons, it prints nothing because the advertisement callback is never called.

@fvincenzo

fvincenzo, 2016-12-12:
Adding @abhishek-pandit @mmorenobarm to follow up.

abhishek-pandit, 2016-12-12:
If this is related to Wicentric stack then @pan- what would you like to be investigated?!

pan-, 2016-12-12:
@abhishek-pandit Well if scanning doesn't work (seems new!), it would be interesting to understand why and resolve it if the issue is caused by the port of BLE API.

pan-, 2017-03-09:
@abhishek-pandit any news on that issue ?

closed 2017-10-27 by ChiefBureaucraticOfficer