ARMmbed/mbed-os #3404: Sporadic semaphore timing issue on NCS36510

geky, 2016-12-08:
The event queue tests have been sporadically picking up an error of 65 milliseconds in the semaphore wait call with the NCS36510.

Here is one example of the failure in https://github.com/ARMmbed/mbed-os/pull/3368#issuecomment-265859831:
```
[1481229975.74][CONN][RXD] >>> Running case #8: 'Testing call_every'...
[1481229975.78][CONN][INF] found KV pair in stream: {{__testcase_start;Testing call_every}}, queued...
[1481229977.24][CONN][RXD] :73::FAIL: Values Not Within Delta 2 Expected 1400 Was 1465
[1481229977.29][CONN][INF] found KV pair in stream: {{__testcase_finish;Testing call_every;0;1}}, queued...
[1481229977.37][CONN][RXD] >>> 'Testing call_every': 0 passed, 1 failed with reason 'Assertion Failed'
```

Type: Bug
Priority: Minor
Target: NCS36510
Toolchain: At least IAR
mbed OS sha: 1b019c3

cc @bridadan

geky, 2016-12-13:
I put together a test that narrows down the bug to an issue with semaphore timing:
https://github.com/geky/mbed/blob/ncs36510-timing/TESTS/events/timing/main.cpp

Unfortunately it does take ~2 hours for issues to crop up, and seems to be non-deterministic...

bridadan, 2016-12-13:
cc @maclobdell @pradeep-gr @radhika-raghavendran

geky, 2016-12-15:
Looks like @bridadan's intuition was correct, the Timer class itself is the problem. Narrowed down the tests so they complete much faster now:
https://github.com/geky/mbed/blob/ncs36510-timing/TESTS/events/timing/main.cpp

It looks like the issue may be in this function:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_ONSEMI/TARGET_NCS36510/ncs36510_us_ticker_api.c#L109

maclobdell, 2017-01-10:
@geky is the issue the semaphore timing (which I assume uses ticks generated by systick interrupt) or is the issue that the test uses us_ticker_read to check the semaphore timing?  In the failing case, is it consistently 65ms  shorter than expected.  @radhika-raghavendran  @pradeep-gr  - is there any possible condition that could sometimes cause us_ticker_read function to return a value that is smaller than expected?

maclobdell, 2017-02-07:
@geky  Please share more information about this issue.

geky, 2017-02-15:
Sorry about the delay, https://github.com/ARMmbed/mbed-os/pull/3779 should be able to fix this

closed 2017-02-21 by sg-

0xc0170, 2017-05-26:
We came across a failure today again with the exact same problem. As it used to be , seems correct (use 2 timers, one is counting msb, the other one is executing ticker interrupts). The fix proposed here and that was integrated looks fine, just doing it a bit differently (not completely certain how it fixes it,).

One thing I noticed the timer disables just own interrupt, which is fine to protect mbs counter, but could potentially caused drifts if other interrupts are involved ?

Let's keep an eye on CI , and see if this becomes real issue or was just one time failure.

pradeep-gr, 2017-05-26:
cc @danish-iftikhar @danclement @jacobjohnson-ON

0xc0170, 2017-05-26:
I should have added this improvement https://github.com/ARMmbed/mbed-os/pull/4094 to my previous comment. this might fix it as it will work even with 16bit timer (no need to have 32bit us ticker).