ARMmbed/mbed-os #3406: OOB: Beetle exporter for uvision5 not working with mbed-os-example-blinky

mikaleppanen, 2016-12-09:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
Beetle (ARM_BEETLE_SOC)

**Toolchain:**
uvision5 exporter

**mbed-cli version:**
1.0.0.

**Expected behavior**
Exporters work

**Actual behavior**
following error "ARM_BEETLE_SOC not supported by uvision5":

-----------------
C:\GIT\oob\mbed-os-example-blinky>mbed export -i uvision5 -m ARM_BEETLE_SOC
C:\Python27\lib\site-packages\fuzzywuzzy\fuzz.py:35: UserWarning: Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning
  warnings.warn('Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning')
usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S] [-E] [--source SOURCE_DIR] [-D MACROS]
                  [--profile PROFILE] [--update-packs]
project.py: error: ARM_BEETLE_SOC not supported by uvision5
[mbed] ERROR: "python" returned error code 2.
[mbed] ERROR: Command "python -u C:\GIT\oob\mbed-os-example-blinky\mbed-os\tools\project.py -i uvision5 -m ARM_BEETLE_SOC --source ." in "C:\GIT\oob\mbed-os-example-blinky"
----------------

**Steps to reproduce**
import mbed-os-example-blinky
try to export using "mbed export -i uvision5 -m ARM_BEETLE_SOC"
----------------------------------------------------------------

mikaleppanen, 2016-12-09:
Oops. wrong version of the mbed-os. Works ok. closing this.

mikaleppanen, 2016-12-09:
Re-opend, happens with 5.3.0-rc3 also.

bridadan, 2016-12-09:
I think this might be related to the fact that the Beetle requires a beta CMSIS pack to be installed. @sarahmarshy does that sound right or have exporters been disabled for this target?

0xc0170, 2016-12-12:
> I think this might be related to the fact that the Beetle requires a beta CMSIS pack to be installed. @sarahmarshy does that sound right or have exporters been disabled for this target?

How can I get a list of targets that are supported for uvision5? I would like to get a list of all targets that are supported for a tool. In this case, uvision5.

theotherjimmy, 2016-12-12:
There is a matrix that you can check (I'm sorry that it can be hard to read) `mbed export -S`