ARMmbed/mbed-os #3407: Support hex file drag-n-drop flash?

ccli8, 2016-12-09:
### Description
- Type: Question
- Priority: Minor

## Question
According to the doc **mbed Interface Firmware Requirements**, ICE firmware needs to support binary/hex file drag-n-drop flash. Currently, I just see binary file flash, but no hex file flash. Is support for hex file flash a must? Where would hex file flash be used?

0xc0170, 2016-12-09:
cc @sg- @c1728p9

0xc0170, 2016-12-09:
I believe it's binary or hex files support (not a must for both). The above persons to confirm.

bridadan, 2017-01-17:
@c1728p9 @sg- bump!

sg-, 2017-01-18:
Yes hex file support must be supported https://www.mbed.com/en/about-mbed/mbed-enabled/mbed-enabled-program-requirements/

This will become important as applications have encoded start addresses where things like a bootloader are present

closed 2017-01-18 by sg-