ARMmbed/mbed-os #3412: export NUCLEO_F746ZG in IAR does not work, probably due to targets.py or targets.json

adustm, 2016-12-09:
### Description
- Type: Bug
- Priority: Blocker
---------------------------------------------------------------
## Bug
Trying to debug issue #3387, I am blocked by the fact that I cannot export for NUCLEO_F746ZG, while it works fine for NUCLEO_F429ZI

**Target**
NUCLEO_F746ZG
**Toolchain:**
IAR
**Toolchain version:**
IAR 7.50.3.10751

**mbed-cli version:** 
0.8.2

**meed-os sha:**
7b974ca Merge pull request #3394 from ARMmbed/release-candidate (mbed-os-5.3.0-rc3)

**Expected behavior**
`mbed export -i iar -m nucleo_f746zg` generates an IAR project

**Actual behavior**
```
mbed export -i iar -m nucleo_f746zg
usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S] [-E] [--source SOURCE_DIR] [-D MACROS]
                  [--profile PROFILE] [--update-packs]
project.py: error: NUCLEO_F746ZG not supported by iar
[mbed] ERROR: "python" returned error code 2.
[mbed] ERROR: Command "python -u C:\MCDGitW601\mbed\tools\project.py -i iar -m nucleo_f746zg --source ." in "C:\MCDGitW601\mbed"
```
I have started the investigation. Here is what I have found : 
The line `exporter, toolchain_name = get_exporter_toolchain(options.ide)` in [mbed-os/tools/project.py](https://github.com/ARMmbed/mbed-os/blob/master/tools/project.py#L237)
returns a list that does not contain NUCLEO_F746ZG
```
['DISCO_L053C8', 'OC_MBUINO', 'NUCLEO_F302R8', 'KL05Z', 'NUCLEO_F334R8', 'NUCLEO_F411RE', 'ARCH_PRO', 'K20D50M', 'NUCLEO_L031K6', 'KL43Z', 'K22F', 'NUCLEO_F401RE', 'NUCLEO_F303RE',
 'LPC11U24', 'UBLOX_C027', 'NUCLEO_F429ZI', 'NUCLEO_F091RC', 'TY51822R3', 'DELTA_DFBM_NQ620', 'MTS_MDOT_F405RG', 'EFM32HG_STK3400', 'EFM32LG_STK3600', 'XBED_LPC1768', 'NUCLEO_L152R
E', 'NUCLEO_F439ZI', 'UBLOX_EVK_ODIN_W2', 'LPC812', 'NUCLEO_F070RB', 'NUCLEO_F072RB', 'MTS_DRAGONFLY_F411RE', 'EFM32WG_STK3800', 'NUCLEO_F446ZE', 'HEXIWEAR', 'NUCLEO_F446RE', 'NUCL
EO_F042K6', 'NUCLEO_F103RB', 'EFM32ZG_STK3200', 'EFM32GG_STK3700', 'MTS_MDOT_F411RE', 'NUCLEO_F030R8', 'NUCLEO_F303K8', 'K64F', 'DISCO_F429ZI', 'NUCLEO_L073RZ', 'NRF52_DK', 'KL46Z'
, 'NRF51_DK', 'KL25Z', 'NUCLEO_F031K6', 'LPC1768']

```
Looking deeper at the scripts, I found that the content of TARGETS in [targets.py](https://github.com/ARMmbed/mbed-os/blob/master/tools/targets.py#L499) is not correct for NUCLEO_F746ZG. **I have the feeling that the targets.json parser cannot manage to parse NUCLEO_F746ZG datas correctly**
```
Target(name='NUCLEO_F429ZI', json_data={'NUCLEO_F429ZI': OrderedDict([('supported_form_factors', ['ARDUINO']), ('inherits', ['Target']), ('core', 'Cortex-M4F'), ('default_toolchain', 'ARM'), ('extra_labels', ['STM', 'STM32F4', 'STM32F429', 'STM32F429ZI', 'STM32F429xx', 'F429_F439']), ('supported_toolchains', ['ARM', 'uARM', 'GCC_ARM', 'IAR']), ('progen', OrderedDict([('target', 'nucleo-f429zi')])), ('macros', ['RTC_LSI=1', 'TRANSACTION_QUEUE_SIZE_SPI=2']), ('device_has', ['ANALOGIN', 'ANALOGOUT', 'CAN', 'ERROR_RED', 'I2C', 'I2CSLAVE', 'I2C_ASYNCH', 'INTERRUPTIN', 'LOWPOWERTIMER', 'PORTIN', 'PORTINOUT', 'PORTOUT', 'PWMOUT', 'RTC', 'SERIAL', 'SERIAL_FC', 'SLEEP', 'SPI', 'SPISLAVE', 'SPI_ASYNCH', 'STDIO_MESSAGES', 'TRNG']), ('detect_code', ['0796']), ('features', ['LWIP']), ('release_versions', ['2', '5']), ('device_name', 'STM32F429ZI')]), 'Target': OrderedDict([('core', None), ('default_toolchain', 'ARM'), ('supported_toolchains', None), ('extra_labels', []), ('is_disk_virtual', False), ('macros', []), ('device_has', []), ('features', []), ('detect_code', []), ('public', False), ('default_lib', 'std')])}, resolution_order=[('NUCLEO_F429ZI', 0), ('Target', 1)], resolution_order_names=['NUCLEO_F429ZI', 'Target'])
Target(name='NUCLEO_F746ZG', json_data={'Target': OrderedDict([('core', None), ('default_toolchain', 'ARM'), ('supported_toolchains', None), ('extra_labels', []), ('is_disk_virtual', False), ('macros', []), ('device_has', []), ('features', []), ('detect_code', []), ('public', False), ('default_lib', 'std')]), 'NUCLEO_F746ZG': OrderedDict([('supported_form_factors', ['ARDUINO']), ('inherits', ['Target']), ('core', 'Cortex-M7F'), ('default_toolchain', 'ARM'), ('extra_labels', ['STM', 'STM32F7', 'STM32F746', 'STM32F746ZG', 'F746_F756']), ('supported_toolchains', ['ARM', 'uARM', 'GCC_ARM', 'IAR']), ('progen', OrderedDict([('target', 'nucleo-f746zg')])), ('macros', ['TRANSACTION_QUEUE_SIZE_SPI=2']), ('device_has', ['ANALOGIN', 'ANALOGOUT', 'CAN', 'I2C', 'I2CSLAVE', 'INTERRUPTIN', 'LOWPOWERTIMER', 'PORTIN', 'PORTINOUT', 'PORTOUT', 'PWMOUT', 'RTC', 'SERIAL', 'SERIAL_ASYNCH', 'SLEEP', 'SPI', 'SPISLAVE', 'SPI_ASYNCH', 'STDIO_MESSAGES', 'TRNG']), ('detect_code', ['0816']), ('features', ['LWIP']), ('release_versions', ['2', '5']), ('device_name', 'STM32F746ZG')])}, resolution_order=[('NUCLEO_F746ZG', 0), ('Target', 1)], resolution_order_names=['NUCLEO_F746ZG', 'Target'])

```
 but I don't know how the json files work, I cannot help on this. Could anyone look please ? 

**Steps to reproduce**
You can use my debug branch https://github.com/adustm/mbed/tree/f746zg_iar_debug if you want to see my debug printf logs

`mbed export -i iar -m nucleo_f746zg`

theotherjimmy, 2016-12-09:
@adustm Have you tried correcting the targets.json entry for the F746ZG? did that work?

adustm, 2016-12-09:
Hi,
You can check in my branch, I've added the progen line that was previously missing.
I've tried to put everything in the same order than for NUCLEO_F429ZI (my _ok_ reference)
With no success :(

theotherjimmy, 2016-12-09:
progen? That's not used.

theotherjimmy, 2016-12-09:
it looks like there is a typo in the IAR definitions here: https://github.com/ARMmbed/mbed-os/blob/master/tools/export/iar/iar_definitions.json#L74

theotherjimmy, 2016-12-09:
I think it will work for you if you upcase the key and remove the extra F

theotherjimmy, 2016-12-09:
Hey, PR up that fixes the issue #3413 @adustm 

Thanks for doing most of the investigation for me

adustm, 2016-12-09:
Thank you, it works now (well at least not I can try to debug my failing project :) )
You say that progen is no more used ? Even in mbed2 environment ?
It is still present for NUCLEO_F429ZI...

Do you want me to close this issue or wait until your PR is merged ?

bridadan, 2016-12-09:
Let's wait until #3413 is merged before closing this. Thanks for tackling this @theotherjimmy and @adustm!

theotherjimmy, 2016-12-09:
>You say that progen is no more used ? Even in mbed2 environment ?

That will be true when the tools used here are released for the mbed2 environment.

>It is still present for NUCLEO_F429ZI...

It won't harm anything to have them in there. Except that it might confuse potential contributers.

closed 2016-12-15 by sg-

adustm, 2016-12-15:
OK

PR #3413 (2016-12-15): Deduplicate IAR exporter templates and enable a few more targets
## Description
Fixes a typo in the IAR definitions dict. should resolve #3412 

## Status
**READY**

## Todos
- [x] export-build CI

## Steps to test or reproduce
`mbed export -i iar -m nucleo_f746zg` from any mbed project would tell you it's not supported, when it could have been.