ARMmbed/mbed-os #3418: NUMAKER_PFM_NUC472 does not work with uvision

c1728p9, 2016-12-09:
When the NUMAKER_PFM_NUC472 is exported and built form uvision there is an error saying that IPv4 or IPv6 must be enabled.

c1728p9, 2016-12-09:
CC: @theotherjimmy

cyliangtw, 2016-12-12:
Please ensure your Keil version MDK 5.x firstly.
Cause current online IDE lost "-DDEVICE_TRNG=1" in export feature's project generation, you could add "-DDEVICE_TRNG=1" in Keil project C/C++ "Misc Controlls" setting by manual.
After that, if you still can't resolve, please guide me how to reproduce it.

bridadan, 2016-12-13:
We're currently running exporter tests on this platform in CI. The following examples all export and build correctly in uVision (as far as the CI system is concerned):

```
mbed-os-example-mesh-minimal
mbed-os-example-blinky
mbed-os-example-sockets
```

@c1728p9 Was this for a particular example?

c1728p9, 2016-12-14:
This came up for @theotherjimmy during 5.3 RC2 out of box testing on the blinky example - "Uvision5 failure: #error directive: "Either IPv4 or IPv6 must be enabled".

cyliangtw, 2016-12-17:
We verified Export Uvision5 prj function of mbed-cli is OK with mbed-os-5.3.0-rc2 .
Maybe, it's online IDE export feature issue.

theotherjimmy, 2016-12-19:
Yup it seems to be fixed. I can't reproduce it anymore.

closed 2016-12-19 by theotherjimmy