ARMmbed/mbed-os #3419: Odin ethernet does not work without configuration

c1728p9, 2016-12-09:
This may be the intended behavior but Ethernet on the Odin does not work without configuration in targets.json since wifi is the default network interface.

c1728p9, 2016-12-09:
CC: @geky, @bulislaw

geky, 2016-12-09:
Is this related to https://github.com/ARMmbed/mbed-os/issues/3269?

c1728p9, 2016-12-09:
Making that change would fix this. This is just to track the OOB failure.

bulislaw, 2016-12-12:
Right now it's either Ethernet or WiFi, having WiFi on by default was requested by SamG. Can we close it?

0xc0170, 2016-12-12:
> Right now it's either Ethernet or WiFi, having WiFi on by default was requested by SamG. Can we close it?

This is known-limitation. @bulislaw is this documented anywhere, and how to enable ethernet (remove EMAC macro definition without touching targets.json file?) ? That would resolve this:

> Making that change would fix this. This is just to track the OOB failure.

bulislaw, 2016-12-12:
It's not. I'm guessing having it on the board page would be good.

bridadan, 2016-12-12:
> I'm guessing having it on the board page would be good.

That'd be great! Or perhaps adding a target configuration option for this? https://github.com/ARMmbed/mbed-os/blob/master/docs/config_system.md#configuration-data-in-targets

bulislaw, 2016-12-13:
@andreaslarssonublox could you please add a note about Odin supporting WiFi by default (and that users need to remove EMAC from the targets to get Ethernet working) https://developer.mbed.org/platforms/ublox-EVK-ODIN-W2/ hopefully we can add proper networking support soon and have both at the same time.

andreaslarssonublox, 2016-12-13:
@bulislaw Yes, I will add some note on the page https://developer.mbed.org/platforms/ublox-EVK-ODIN-W2

bulislaw, 2016-12-13:
Awesome, thanks!

closed 2016-12-13 by bulislaw

bridadan, 2016-12-13:
@andreaslarssonublox Thanks for adding that!

geky, 2016-12-13:
Updated mbed-os-example-sockets to automatically use the ethernet configuration, the wifi configuration is available as part of mbed-os-example-wifi: