ARMmbed/mbed-os #3420: Can not wakeup from deepsleep with LowPowerTicker?

maskedw, 2016-12-10:
### Description
- Type: Question

---------------------------------------------------------------
## Question 

**Target**
NECLEO_L432KC

**Toolchain:**
GCC_ARM

**Toolchain version:**
```
$ arm-none-eabi-g++ --version
arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
```

**mbed-cli version:** 
```
$ mbed --version
0.9.10
```

**meed-os sha:**
```
$ git log -n1 --oneline
c57427f Merge pull request #3321 from jeromecoutant/PR_L432KC
```

**Expected behavior**

Wakeup from "deepSleep" with "LowPowerTicker ISR"

**Actual behavior**

It does not wakeup from deepsleep and ISR is not called.

**Steps to reproduce**

```
    #include "mbed.h"
    
    Serial pc(USBTX, USBRX);
    LowPowerTicker ticker;
    
    void callback()
    {
    }
    
    int main(int argc, char const* argv[])
    {
        pc.baud(115200);
        pc.printf("Hello mbed\n");
    
        ticker.attach(callback, 1);
        for (;;)
        {
            pc.printf("Do deepsleep\n");
            wait_ms(500);
            deepsleep();
            pc.printf("Wakeup deepsleep\n");
            wait_ms(500);
        }
        return 0;
    }
```

-----------------------------------------------------------------

## Question

**How to?**
Can not wakeup from deepsleep with LowPowerTicker?

jeromecoutant, 2016-12-12:
Hi
I agree that LowPower timer/ticker currently doesn't work correctly for STM32L4 targets...
I am working on it
Regards,

maskedw, 2016-12-13:
I stopped using LSE and decided to use LSI (-DRTC_LSI = 1).
Then LowPowerTicker worked.

jeromecoutant, 2017-07-13:
Hi
Should we close this issue ?
I made few quick tests, and it seems that it works with LSE and LSI ?

maskedw, 2017-07-30:
I'm sorry. There is no environment that can be confirmed.
If there is no problem in operation, please close it.

0xc0170, 2017-08-03:
Please reopen with more current information if this issue is still present.

closed 2017-08-03 by 0xc0170