ARMmbed/mbed-os #3425: LowPowerTimer maximum value is only 1 hour

jeromecoutant, 2016-12-12:
Hi

I understood that LowPowerTimer resolution is microseconds due to compatibility reason.

This means that maximum timer duration is 0xFFFFFFFF microseconds, so about 1 hour.
For me, LowPowerTimer should be much longer than 1 hour...

Can you share your point of view about LowPowerTimer  feature?

Thx

bridadan, 2016-12-12:
cc @0xc0170

0xc0170, 2016-12-13:
> For me, LowPowerTimer should be much longer than 1 hour...

Ideally yes. The resolution could be milliseconds.
I would rather have 64 bit width which would then extend this to years. Same could be applied to us ticker API. I recall there was this change done but failed to cover some cases, caused some problems and was reverted. I would need to have a look through the git log to find out more.

bridadan, 2016-12-13:
@maclobdell you may have some thoughts on this as well?

jeromecoutant, 2017-09-26:
Hi
Back to this old issue.
It seems that resolution is now u64. Is that correct ?

0xc0170, 2017-09-26:
> It seems that resolution is now u64. Is that correct ?

Yes, as Ticker uses `insert_absolute` that has `us_timestamp_t` as argument. cc @pan-

pan-, 2017-09-26:
Yes, resolution is `uint64_t` now.

jeromecoutant, 2017-09-26:
Maybe we should update hal files updating timestamp_t to us_timestamp_t ?

0xc0170, 2017-09-26:
> Maybe we should update hal files updating timestamp_t to us_timestamp_t ?

Why would be the benefit? In most cases, the hw counter is 32bit wide

jeromecoutant, 2017-09-26:
OK, I understand.
I can see attach_us is well using us_timestamp_t

I think you can close this issue ?

closed 2017-09-26 by 0xc0170