ARMmbed/mbed-os #3426: TimeStamp starts from 0 after microcontroller wakeup from Standby mode

ArjunSinghh, 2016-12-12:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug | Enhancement | Question

I am using STM32l06 microcntroller's custom board and when microcontroller wakesup from stand by mode, according to datasheet, it should not loose time stamp, but for me it always start from 0. The reason I found is that in rtc_api.c file  there was a flag that  gets cleared every time after controller comes out of standby mode, and in mbed library, the code is written to check if that flag is cleared then start time stamp from 0 again, now I dont know if it is right approach or not, but in mbed file i made change, i checked if rtc_read gives a value greater than some significant value, then i return 1, so time doesnt start from 0 again. So the problem is fixed, but I had to make changes in mbed file,. seems like mbed need to update that file.

jeromecoutant, 2016-12-12:
Hi
Please propose your patch
Thx

janjongboom, 2016-12-12:
FYI, from my conversation with @ArjunSinghh:

>  Actually there was a flag that  gets cleared every time after controller comes out of standby mode, and in mbed library, the code is written to check if that flag is cleared then start time stamp from 0 again, now I dont know if it is right approach or not, but in mbed file i made change, i checked if rtc_read gives a value greater than some significant value, then i return 1, so time doesnt start from 0 again. SO the problem is fixed, but I had to make changes in mbed file,. seems like mbed need to update that file.

This should be fixed in the HAL for STM32l06

jeromecoutant, 2017-05-22:
Hi @ArjunSinghh 
Is issue still present?
Can you give us some code to reproduce it ?
Thx

jeromecoutant, 2017-07-13:
As there is no feedback, should we close this issue ?
Maybe behavior is better after #3743 patch ?

jeromecoutant, 2017-09-26:
ST_TO_BE_CLOSED

closed 2017-09-27 by 0xc0170