ARMmbed/mbed-os #3435: weird return value of __svcMutexWait/__svcMutexRelease

kyliuxing, 2016-12-13:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type:  Question
- Related issue: 
- Priority: Blocker 

---------------------------------------------------------------

## Question
I feel the return of __svcMutexWait/__svcMutexRelease should be  the value in enum list of osStatus, but sometimes it will return the value of mutex_id.
So my question is when the function will return the mutex_id as result?Is it normal?
It is a always issue when we use lwip stack to download and has blocked our developing,  so please give some advices.
Thank you!

RDA shileiyu

0xc0170, 2016-12-13:
Hi,

can you provide code reference about return values? That would be helpful. (mainly where is muted_id returned). I could not find it.

cc @c1728p9 @bulislaw

bulislaw, 2016-12-13:
I don't see how it could happen -> https://github.com/ARMmbed/mbed-os/blob/master/rtos/rtx/TARGET_CORTEX_M/rt_Mutex.c#L218
can you provide more details and/or example.

kyliuxing, 2016-12-13:
i reworked function to debug as below, osMutexRelease is same , please help check, thank you!
code:
osStatus osMutexWait (osMutexId mutex_id, uint32_t millisec) {
  osStatus res;
  if (__get_IPSR() != 0U) {
    return osErrorISR;                          // Not allowed in ISR
  }

  res = __svcMutexWait(mutex_id, millisec);
  if(res != osOK)
    printf("w %x, f %x\r\n", res, ((P_MUCB)mutex_id)->cb_type);
  return res;
}
output:
w 1827f8, f 3

bulislaw, 2016-12-13:
Worth asking people that develop RTOS https://github.com/ARM-software/CMSIS

0xc0170, 2016-12-15:
Thanks @bulislaw for the pointer.

The question is redirected here https://github.com/ARM-software/CMSIS/issues/103 , closing this issue.

closed 2016-12-15 by 0xc0170