ARMmbed/mbed-os #3441: NUMAKER_PFM_NUC472 LWIP implementation contains printf

bridadan, 2016-12-13:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NUMAKER_PFM_NUC472

**meed-os sha:**
aa6780d Merge pull request #3416 from bridadan/rename_stm32f1_i2c_api

**Notes**
The LWIP implementation for the NUMAKER_PFM_NUC472 currently contains calls to `printf`. These should be removed or repleace with a debug print function that can be enabled conditionally (not sure which option would be preferred here @geky?)

cc @ccli8

geky, 2016-12-13:
Agreed. Perhaps they should be hidden behind an EMAC_DEBUG flag?

bridadan, 2016-12-13:
@geky, or maybe just use `LWIP_DEBUGF` instead of printf? Is that the print macro/function that by default is disabled?

geky, 2016-12-13:
@bridadan `LWIP_DEBUGF` is a good idea, however we are planning on separating the emac implementations from lwip to open up more options with other network stacks. We could use `LWIP_DEBUGF` as a temporary solution until the emac api (which will probably have its own `EMAC_DEBUG` flag) becomes standardized.

bridadan, 2016-12-13:
@geky So moving forward, maybe replace the current printfs with the `LWIP_DEBUGF`, then when you do the emac overhaul they can be changed along with the rest of the `LWIP_DEBUGF` calls?

ccli8, 2016-12-14:
cc @cyliangtw

cyliangtw, 2016-12-14:
@bridadan 
The printf(...) is for log info. Such like, as while you boot linux system, you could also see driver init info including ethernet.
Adopt your suggestion to fix this issue by LWIP_DEBUGF in PR #3434 .

0xc0170, 2016-12-14:
Thanks @cyliangtw @ccli8 for looking at this and fixing it. we will review the PR with fixes soon.

cyliangtw, 2016-12-17:
@0xc0170 
PR #3434 was accepted and merged.
Please help to close it.

closed 2016-12-17 by 0xc0170