ARMmbed/mbed-os #3444: CFSTORE: SRAM mode not working

simonqhughes, 2016-12-14:
Sarah Marsh reported this issue as follows:

From: Sarah Marsh 
Sent: 13 December 2016 23:48
To: Simon Hughes; Marcus Chang
Subject: Enabling config store in SRAM for cloud client

Hello Simon and Marcus,

I have been tasked with getting a cellular device to connect to mbed cloud. It requires configuration store, but I was told there is a way to use config store in SRAM rather than nonvolatile memory for targets that have not implemented a storage class. Marcus suggested using these macros https://github.com/ARMmbed/mbed-os/blob/master/features/storage/FEATURE_STORAGE/cfstore/source/cfstore_config.h#L44-L57 for such a target. Using these and turning on the "STORAGE" feature for my target in targets.json, compilation fails at the reference to cfstore_svm_storage_drv
here:
https://github.com/ARMmbed/mbed-os/blob/754bd243010732d273f5ec931f53945edc2f5405/features/storage/FEATURE_STORAGE/cfstore/source/cfstore_svm.cpp#L63

The macros enable definition of that variable here:
https://github.com/ARMmbed/mbed-os/blob/master/features/storage/FEATURE_STORAGE/cfstore/source/cfstore_svm.cpp#L40-L42. 
Obviously, that is a storage implementation unique to K64F. Should that variable be defined in an else clause for targets without a storage class? Or is there another step I need to take? 

Marcus, sorry for the duplicate information and yet another email, but I wanted to put the information in one place.

Thanks for your time,
Sarah Marsh

simonqhughes, 2016-12-14:
@sarahmarshy @marcuschangarm 

Hi

The following pull request has been submitted which fixes this problem:

https://github.com/ARMmbed/mbed-os/pull/3368

The PR is blocked dues to some unrelated CI intermittent failure (@geky @bridadan  have more details) .

To enable SRAM mode I changed cfstore_config.h to define as follows:

```
#define CFSTORE_STORAGE_DISABLE 1
#if defined DEVICE_STORAGE && CFSTORE_STORAGE_DISABLE==0
#define CFSTORE_CONFIG_BACKEND_FLASH_ENABLED
#endif
```

I used this command to build:

```
mbed -v test --compile -t GCC_ARM -m K64F  2>&1 | tee build_tests_cfstore_issue_3444_20161214_1043.txt
```

Here is the build log showing no problems.

Hope this helps
Simon.

[build_tests_cfstore_issue_3444_20161214_1043.txt](https://github.com/ARMmbed/mbed-os/files/651449/build_tests_cfstore_issue_3444_20161214_1043.txt)

sarahmarshy, 2016-12-14:
Merging that PR and using this macro definition resolved this issue. Thank you.

simonqhughes, 2016-12-14:
Issue resolved so closing.

closed 2016-12-14 by simonqhughes