ARMmbed/mbed-os #3446: Mbed build broken for TARGET UBLOX EVK ODIN W2 using GCC ARM toolchain

haseeb-ublox, 2016-12-14:
Description

Hi,

I have been trying to build mbed os 5 for UBLOX_EVK_ODIN_W2 using Mbed native tools i.e build.py present under mbed-os/tools folder but I am facing the issue while the compiler tries to build wifi_emac_api.py . 
I have used the following command to build:

**>python build.py -m UBLOX_EVK_ODIN_W2 -t GCC_ARM**

I have enabled the verbose option using -v flag and DEBUG trace shows this error:

**[DEBUG] Output: mbed-os/BUILD/mbed/TARGET_UBLOX_EVK_ODIN_W2/TARGET_STM/TARGET_STM32F4/TARGET_UBLOX_EVK_ODIN_W2/sdk/ublox-odin-w2-drivers/cb_main.h:28:25: fatal error: mbed_events.h: No such file or directory**

I am facing this error after pulling latest mbed os this morning. 

- Type: Bug

- Priority: Blocker

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
GCC_ARM

0xc0170, 2016-12-14:
@haseeb-ublox ODIN is only supported for mbed OS 5, therefore use mbed-cli to build an app.

cc @theotherjimmy - might have a look how we can produce more clear message if someone touches scripts inside tools.

theotherjimmy, 2016-12-16:
@haseeb-ublox please use `mbed compile -t gcc_arm -m ublox_evk_odin_w2` to compile. There are a few extra arguments that mbed infers from the environment it's being run in.

@0xc0170 That's a compiler message. I can't easily improve those, escpially since we are locked into GCC version 4.9.3.

0xc0170, 2016-12-19:
> @0xc0170 That's a compiler message. I can't easily improve those, especially since we are locked into GCC version 4.9.3.

my point was about having target only for mbed 5 and using tools/scripts, not about the error message shown. The best might be to have clear documentation mbed-cli vs mbed/tools.

@haseeb-ublox Why did you decide to use tools/build.py and not mbed-cli? Trying to find out if there's lack of documentation or an error somewhere?

theotherjimmy, 2016-12-19:
@0xc0170 So given that when you run `mbed compile` it eventually runs `python tools/build.py` You're asking for an error message when you try to do a build for a target that does not have a `release_version` that the compile would expect? That is something that I could do.

closed 2017-02-04 by sg-