ARMmbed/mbed-os #3447: Debug build with -o "debug-info" failed

ccli8, 2016-12-15:
### Description
- Type: Bug
- Priority: Major

I would like to build debug version but fails with the error log.
```
$ mbed compile -o "debug-info"
usage: make.py [-h] [-m MCU] [-t TOOLCHAIN] [--color] [--cflags CFLAGS]
               [--asmflags ASMFLAGS] [--ldflags LDFLAGS] [-c]
               [--profile PROFILE] [--app-config APP_CONFIG] [-p PROGRAM]
               [-n PROGRAM] [-j JOBS] [-v] [--silent] [-D MACROS] [-S]
               [-f GENERAL_FILTER_REGEX] [--automated] [--host HOST_TEST]
               [--extra EXTRA] [--peripherals PERIPHERALS]
               [--dep DEPENDENCIES] [--source SOURCE_DIR]
               [--duration DURATION] [--build BUILD_DIR] [-N ARTIFACT_NAME]
               [-d DISK] [-s SERIAL] [-b BAUD] [-L] [--rtos] [--rpc] [--eth]
               [--usb_host] [--usb] [--dsp] [--fat] [--ublox] [--testlib]
               [-l LINKER_SCRIPT]
make.py: error: unrecognized arguments: -o debug-info
[mbed] ERROR: "python" returned error code 2.
[mbed] ERROR: Command "python -u C:\ccli8\iot\mbed\mbed-os\nu-testtest2\mbed-os\tools\make.py -t ARM -m NUMAKER_PFM_NUC472 --source . --build .\BUILD\NUMAKER_PFM_NUC472\ARM -o debug-info" in "C:\ccli8\iot\mbed\mbed-os\nu-testtest2"
---
```

---------------------------------------------------------------
## Bug

**Target**
NUMAKER_PFM_NUC472

**Toolchain:**
GCC_ARM|ARM

**mbed-cli version:** 
1.0.0

**meed-os sha:**
mbed-os-5.3.0-rc3

pan-, 2016-12-15:
@ccli8 The flag `-o debug-info` has been replaced by the debug profile: `--profile debug`. 
You can find more information about profiles [here](https://github.com/ARMmbed/mbed-os/blob/master/docs/build_profiles.md).

0xc0170, 2016-12-15:
Note to @theotherjimmy - we shall consider using deprecated route for any public argument changes (as we did for some recently).

@ccli8 If you are having problems any with profiles let us know

ccli8, 2016-12-19:
@pan- @0xc0170 I am adding uvisor support for our new target `NUMAKER_PFM_M487` and find uvisor doc e.g. https://github.com/ARMmbed/uvisor/blob/master/docs/core/DEVELOPING_LOCALLY.md and https://github.com/ARMmbed/uvisor/blob/master/docs/api/DEBUGGING.md don't reflect the change. Besides, I want to know by which the tool adds TARGET_RELEASE or TARGET_DEBUG into compile list. This would help for my porting.

0xc0170, 2016-12-19:
cc @AlessandroA Please check the above, about uvisor docs and debug profile.

AlessandroA, 2016-12-19:
Thank you for reporting this! I created a fix: https://github.com/ARMmbed/uvisor/pull/385

bridadan, 2016-12-19:
Thanks for the fix @AlessandroA! I'll go ahead and close this, thanks for reporting @ccli8!

closed 2016-12-19 by bridadan