ARMmbed/mbed-os #3448: [PATCH] Fix re-entrancy crash bug in mbed_ticker_api.c

bisqwit, 2016-12-15:
```
From 41611c7ed8378ce8a965a8713f45247157388d7f Mon Sep 17 00:00:00 2001
From: Joel Yliluoma <joel.yliluoma@iki.fi>
Date: Thu, 15 Dec 2016 08:43:35 +0200
Subject: [PATCH] Fix crash bug, see long commit message for details

Calling set_interrupt() while the linked list is in broken state
(because not all pointers have been updated) can cause corruption.

This is because set_interrupt() may immediately invoke ticker_irq_handler(),
which, then, calls the event handler, which may call ticker_insert_event()
and ticker_remove_event() functions.
This re-entrancy overrides the critical section supposedly protecting
the manipulation of the linked list.

As a result, the linked list may form infinite loops, and execution
never escapes the nested ticker_insert_event() function.

This patch changes the order of some lines of code to make sure that
the linked list is properly patched up before set_interrupt() is called.
---
 hal/mbed_ticker_api.c | 7 +++++--
 1 file changed, 5 insertions(+), 2 deletions(-)

diff --git a/hal/mbed_ticker_api.c b/hal/mbed_ticker_api.c
index 6e3126d..0faf22c 100644
--- a/hal/mbed_ticker_api.c
+++ b/hal/mbed_ticker_api.c
@@ -75,15 +75,18 @@ void ticker_insert_event(const ticker_data_t *const data, ticker_event_t *obj, t
         prev = p;
         p = p->next;
     }
+    /* if we're at the end p will be NULL, which is correct */
+    obj->next = p;
     /* if prev is NULL we're at the head */
     if (prev == NULL) {
         data->queue->head = obj;
+        /* Note: set_interrupt() may immediately invoke ticker_irq_handler(),
+         *       if timestamp is in the past.
+         */
         data->interface->set_interrupt(timestamp);
     } else {
         prev->next = obj;
     }
-    /* if we're at the end p will be NULL, which is correct */
-    obj->next = p;
 
     core_util_critical_section_exit();
 }
-- 
1.9.1
```

0xc0170, 2016-12-15:
Thanks for the detailed description plus a fix included. Is this worth a small test case to reproduce this corruption and verify the fix?

bisqwit, 2016-12-15:
Sorry, but I am having trouble producing a minimal testcase to trigger the bug. It happened randomly in a program using Semtech's LoRa driver which juggles a number of timeout timers for various purposes and a global timer that triggers at 10ms intervals. Some of those timer callbacks may have taken more than 10ms to finish in some circumstances.

0xc0170, 2016-12-15:
OK thanks for the information. Are you happy to send a patch for review via pull request, or we take an action here?

bisqwit, 2016-12-15:
I don't know how to send pull-requests. You can use git-am to apply the patch from my report.

closed 2017-10-27 by ChiefBureaucraticOfficer