ARMmbed/mbed-os #3463: [STM32] CAN read() return value

szechyjs, 2016-12-16:
### Description
- Type: Bug
- Priority: Major

According to the [CAN.h](https://github.com/ARMmbed/mbed-os/blob/master/drivers/CAN.h#L141-L142) header file the `read()` method is only supposed to return `1` if data was read. All of the STM32 implementations in `can_api.c` are hard coded to `return 1;`.

- [STM32F0](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F0/can_api.c#L270)
- [STM32F1](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F1/can_api.c#L273)
- [STM32F2](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F2/can_api.c#L283)
- [STM32F3](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F3/can_api.c#L273)
- [STM32F4](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/can_api.c#L286)
- [STM32F7](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F7/can_api.c#L286)
- [STM32L4](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F0/can_api.c#L270)

The NXP targets implement this correctly, [can_api.h](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NXP/TARGET_LPC176X/can_api.c#L421-L424)

---------------------------------------------------------------
## Bug

**Target**
STM32

**Expected behavior**
Only return `1` if **new** data was read.

**Actual behavior**
Returns `1` regardless if data was received or not. If a message has previously been received then this will be returned again.

**Steps to reproduce**
```c
CANMessage msg;
while(true)
{
    if (can1.read(msg))
    {
        // This will always print
        printf("rx -id: %i, data: %x\n", msg.id, *msg.data);
    }
}
```

0xc0170, 2016-12-19:
cc @bcostm @adustm @LMESTM @jeromecoutant

Nodraak, 2016-12-21:
Should I submit a PR ? I'll go for a lunch and if nobody says otherwise i'll do it right after :)

commit cefae227 (2017-01-09): Merge pull request #3492 from Nodraak/fix/3463/can_read_return_value

Fix #3463 CAN read() return value
closed 2017-01-09 by sg-