ARMmbed/mbed-os #3464: [STM32] CAN write() return value

szechyjs, 2016-12-16:
### Description
- Type: Bug
- Related issue: #3463
- Priority: Major

According to the [CAN.h](https://github.com/ARMmbed/mbed-os/blob/master/drivers/CAN.h#L130-L131) header file the `write()` method is only supposed to return `1` if send was successful. All of the STM32 implementations in `can_api.c` are hard coded to `return 1;`.

- [STM32F0](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F0/can_api.c#L229)
- [STM32F1](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F1/can_api.c#L232)
- [STM32F2](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F2/can_api.c#L242)
- [STM32F3](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F3/can_api.c#L232)
- [STM32F4](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/can_api.c#L245)
- [STM32F7](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F7/can_api.c#L245)
- [STM32L4](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F0/can_api.c#L232)

The NXP targets implement this correctly, [can_api.h](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NXP/TARGET_LPC176X/can_api.c#L356-L400)

---------------------------------------------------------------
## Bug

**Target**
STM32

**Expected behavior**
Only return `1` if write was successful.

**Actual behavior**
Returns `1` regardless if write was successful.

**Steps to reproduce**
```c
if (can1.write(CANMessage(1337, &data, sizeof(data)))
{
    // This will always print
    printf("tx\n");
}
```

0xc0170, 2016-12-19:
cc @bcostm @adustm @LMESTM @jeromecoutant

adustm, 2016-12-19:
Hello @0xc0170 @szechyjs 
It is not obvious to me that the can_write function is supposed to do some polling as long as the msg is not written.
We are just preparing the sending of the  the system and setting the interruptions to be notified when the msg is sent.

Cheers

adustm, 2017-05-05:
Hello
What do we do with this issue ? Is it sure that we are supposed to wait ?
I've seen that @Nodraak has created a PR for the read part. Would you be interested into creating a PR for the write part also ?

Cheers

Nodraak, 2017-07-06:
Hi !

What is the status here ? It seems that the issue has been fixed by commit 1c4e0d7dcbc78dfa066e64f16213317b8cb91395, as [can_write() can sometimes return 0](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/can_api.c#L249).

adustm, 2017-08-01:
>  It seems that the issue has been fixed by commit 1c4e0d7, as can_write() can sometimes return 0.
I think you are right. 
@szechyjs could you please check if it fits your need and maybe clode the issue ?

Kind regards

0xc0170, 2017-08-02:
I'll close this . Please reopen if any new update

closed 2017-08-02 by 0xc0170