ARMmbed/mbed-os #3495: WONT COMPILE : EthernetInterface/mbed-rtos @ mbed V131, v132

andcor02, 2016-12-22:
So i have had reports that the following EthernetInterface/mbed-rtos does not compile with mbed 2.0 versions from V130 onwards. 

Original issue:

https://developer.mbed.org/questions/76329/Compiler-errors-with-mbed-V131-v132-and-/

I can confirm this is correct.

**To recreate the issue:**

- **Import:** https://developer.mbed.org/teams/NXP/code/Ethernet_UDP_client/?platform=FRDM-K64Fhttps://developer.mbed.org/teams/NXP/code/Ethernet_UDP_client/?platform=FRDM-K64F

- **Select "mbed" revision V131/V132**

- **Receive errors:**

`Error: Struct "_enet_handle" has no field "rxBdDirty" in "EthernetInterface/lwip-eth/arch/TARGET_Freescale/k64f_emac.c", Line: 86, Col: 19`

`Error: Struct "_enet_handle" has no field "rxBdDirty" in "EthernetInterface/lwip-eth/arch/TARGET_Freescale/k64f_emac.c", Line: 89, Col: 19`

Regards,

Andrea

0xc0170, 2016-12-22:
Thanks for the reporting. we shall look at this.

I tracked down the commit that updated KSKD and removed  that structure member : 931da516a974a0f3d3aaefdb47.  This commit needs to be reflected to EthernetInterface - 177eaff8475bf07fda2fded.

cc @adbridge

bridadan, 2017-01-17:
@andcor02 @adbridge any update?

andcor02, 2017-01-18:
@bridadan @adbridge 

Just checked this, updating to v133 fixed the issues. Closing this. Thanks!

andcor02, 2017-01-20:
Just rechecked all of this, will not compile with all libraries updated to revisions:

I get two compiler errors:

mbed.lbr - v133 & v134

mbed-rtos = v123

EthernetInterface = v54

in file: EthernetInterface/lwip-eth/arch/TARGET_Freescale/k64f_emac.c lines 86 and 89 are the problem

bridadan, 2017-01-20:
@andcor02 Do you have a published program you can point us at so we can reproduce it?

andcor02, 2017-01-20:
@bridadan 

To recreate the issue:

Import: https://developer.mbed.org/teams/NXP/code/Ethernet_UDP_client/?platform=FRDM-K64Fhttps://developer.mbed.org/teams/NXP/code/Ethernet_UDP_client/?platform=FRDM-K64F

Update all libraries to latest revision :)

bridadan, 2017-01-20:
@andcor02 thanks, I was able to reproduce! @0xc0170 Any ideas for a fix?

reanovich, 2017-10-26:
I got this problem too when I'm using EthernetInterface library with mbed v154 and mbed-rtos v125
To resolve this issued I have found 2 dumb solutions:
1. Delete those two lines at **k64f_emac.c** which contain **g_handle.rxBdDirty**
2. Use mbed v120 and mbed-rtos v114

the problem occur because there is no **rxBdDirty** in _enet_handle_t struct. You can find this struct initiation [here](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/drivers/fsl_enet.h)

0xc0170, 2018-06-15:
Thanks @reanovich for sharing.  Those 2 are actually good proposals. The rxBdDirty is not used anywhere, can be removed from there with no change to the result.

As these are legacy libraries (updated, reworked network libraries replaced those), we will close this as won't fix.

closed 2018-06-15 by 0xc0170