ARMmbed/mbed-os #3496: The return of malloc is out of the range of heap.

kyliuxing, 2016-12-22:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Priority: Blocker
---------------------------------------------------------------
## Bug
**Target**
RDA 5991H
**Toolchain:**
ARM 5.1.7

At the beginning, malloc works normally, but after a while, i found the return is out of the range of heap. The heap start is 0x180000 but the return pointer is 0x102a80.
so please give some advice to check it, thank you!

shileiyu
RDA

0xc0170, 2016-12-22:
Can you create a code snippet to reproduce the problem?

kyliuxing, 2016-12-23:
I have found root cause, i free a wrong address then it was added to heap.
thanks for your help!

closed 2016-12-23 by kyliuxing