ARMmbed/mbed-os #3498: K64F - CI test shield failures

0xc0170, 2016-12-22:
### Description
- Type: Bug | Question
- Priority: Major

---------------------------------------------------------------
## Bug

Failing some test cases

**Target**
K64F

**Toolchain:**
ARM (tested just ARM)

**Toolchain version:**
ARM 5.17

**mbed-cli version:** 
1.0.0

**meed-os sha:**
f6f872a Merge pull request #3159 from radhika-raghavendran/master

Latest master, only failures captured below:

```
mbedgt: test case report:
+----------+---------------+----------------------------+------------------------------------------------------------+--------+--------+--------+--------------------+
| target   | platform_name | test suite                 | test case                                                  | passed | failed | result | elapsed_time (sec) |
+----------+---------------+----------------------------+------------------------------------------------------------+--------+--------+--------+--------------------+
| K64F-ARM | K64F          | tests-api-analogout        | AnalogOut on AOUT                                          | 0      | 2      | FAIL   | 0.15               |
| K64F-ARM | K64F          | tests-api-businout         | BusInOut to BusInOut                                       | 0      | 6      | FAIL   | 16.42              |
| K64F-ARM | K64F          | tests-api-digitalio        | Digital I/O on DIO_8/DIO_9                                 | 0      | 1      | FAIL   | 0.12               |
| K64F-ARM | K64F          | tests-api-digitalio        | Digital I/O on DIO_9/DIO_8                                 | 0      | 1      | FAIL   | 0.11               |
| K64F-ARM | K64F          | tests-api-i2c              | I2C -  EEProm WR 2 Bytes                                   | 0      | 3      | FAIL   | 0.23               |
| K64F-ARM | K64F          | tests-api-i2c              | I2C -  EEProm WR Single Byte                               | 0      | 2      | FAIL   | 0.21               |

| K64F-ARM | K64F          | tests-api-interruptin      | InterruptIn on DIO_8                                       | 0      | 2      | FAIL   | 0.17               |
| K64F-ARM | K64F          | tests-api-interruptin      | InterruptIn on DIO_9                                       | 0      | 2      | FAIL   | 0.17               |
| K64F-ARM | K64F          | tests-api-pwm              | PWM_0 Frequency 100ms                                      | 0      | 4      | FAIL   | 10.47              |
| K64F-ARM | K64F          | tests-api-pwm              | PWM_1 Frequency 100ms                                      | 0      | 4      | FAIL   | 10.48              |
| K64F-ARM | K64F          | tests-api-pwm              | PWM_3 Duty Cycle 100ms                                     | 0      | 8      | FAIL   | 12.75              |
| K64F-ARM | K64F          | tests-api-pwm              | PWM_3 Duty Cycle 10ms                                      | 0      | 8      | FAIL   | 1.95               |
| K64F-ARM | K64F          | tests-api-pwm              | PWM_3 Duty Cycle 30ms                                      | 0      | 8      | FAIL   | 4.36               |
| K64F-ARM | K64F          | tests-api-pwm              | PWM_3 Frequency 100ms                                      | 0      | 4      | FAIL   | 10.48              |
| K64F-ARM | K64F          | tests-api-pwm              | PWM_3 Frequency 10ms                                       | 0      | 4      | FAIL   | 10.48              |
| K64F-ARM | K64F          | tests-api-pwm              | PWM_3 Frequency 30ms                                       | 0      | 4      | FAIL   | 3.48               |

+----------+---------------+----------------------------+------------------------------------------------------------+--------+--------+--------+--------------------+

```

Analogout can be ignored, I did not wire the pins. PWM probably known issue ? I2C should be tracked here as well, I have seen an issue on ci-test-shield repository (https://github.com/ARMmbed/ci-test-shield).
My board revision is B. We should document somewhere (probably in the CI test shield repository) known limitations of a platform, if there are any.

@BlackstoneEngineering @mmahadevan108

BlackstoneEngineering, 2016-12-23:
I will address these in turn

 ## PWM
On the K64F anything grater than ~63ms fails, so the 10ms and 30ms tests should pass, **these failures are a problem and should be looked into**. 

## Interrupt In / Digital IO on D8/9
Check which hardware version of the K64F you are using, as i recall you tend to use an out of date K64F, we updated the default pins to work on the latest (REVD 3 ?) hardware ~2months ago, so if you have older hadware the mcu pins are different and thus will fail. 

## BusIn/Out
fails because it uses D8/9, see previous explanation above

## I2C
Yup, there is a problem with I2C on the K64F, known problem, see here : https://github.com/ARMmbed/ci-test-shield/issues/17 , @maclobdell is working on it with @mmahadevan108 i believe. 

## AnalogI/O
invalid failure as you didnt hook up the pin.

0xc0170, 2017-02-21:
Update: latest master (sha 7fc73e4)

i2c is fixed,
```
+--------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
| target       | platform_name | test suite    | test case                          | passed | failed | result | elapsed_time (sec) |
+--------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
| K64F-GCC_ARM | K64F          | tests-api-i2c | I2C -  EEProm 2nd WR 10  Bytes     | 1      | 0      | OK     | 0.06               |
| K64F-GCC_ARM | K64F          | tests-api-i2c | I2C -  EEProm 2nd WR 100 Bytes     | 1      | 0      | OK     | 0.08               |
| K64F-GCC_ARM | K64F          | tests-api-i2c | I2C -  EEProm 2nd WR 2 Bytes       | 1      | 0      | OK     | 0.06               |
| K64F-GCC_ARM | K64F          | tests-api-i2c | I2C -  EEProm 2nd WR Single Byte   | 1      | 0      | OK     | 0.07               |
| K64F-GCC_ARM | K64F          | tests-api-i2c | I2C -  EEProm WR 10  Bytes         | 1      | 0      | OK     | 0.06               |
| K64F-GCC_ARM | K64F          | tests-api-i2c | I2C -  EEProm WR 100 Bytes         | 1      | 0      | OK     | 0.07               |
| K64F-GCC_ARM | K64F          | tests-api-i2c | I2C -  EEProm WR 2 Bytes           | 1      | 0      | OK     | 0.06               |
| K64F-GCC_ARM | K64F          | tests-api-i2c | I2C -  EEProm WR Single Byte       | 1      | 0      | OK     | 0.06               |
| K64F-GCC_ARM | K64F          | tests-api-i2c | I2C -  Instantiation of I2C Object | 1      | 0      | OK     | 0.07               |
| K64F-GCC_ARM | K64F          | tests-api-i2c | I2C -  LM75B Temperature Read      | 1      | 0      | OK     | 0.06               |
+--------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
```

The last one to be fixed is PWM.

BlackstoneEngineering, 2017-02-22:
The PWM issue is in 2 parts
1. the K64F cannot go faster than 63.3ms, therefore the 100ms tests will always fail.
2. the PWM3 tests also use pin 8/9, therefore if you have an old board with a different mapping it will fail.

Here are the results from the latest master branch. PWM passes as expected. 

```
+--------------+---------------+---------------+--------+--------------------+-------------+
| target       | platform_name | test suite    | result | elapsed_time (sec) | copy_method |
+--------------+---------------+---------------+--------+--------------------+-------------+
| K64F-GCC_ARM | K64F          | tests-api-pwm | FAIL   | 179.49             | shell       |
+--------------+---------------+---------------+--------+--------------------+-------------+
mbedgt: test suite results: 1 FAIL
mbedgt: test case report:
+--------------+---------------+---------------+------------------------+--------+--------+--------+--------------------+
| target       | platform_name | test suite    | test case              | passed | failed | result | elapsed_time (sec) |
+--------------+---------------+---------------+------------------------+--------+--------+--------+--------------------+
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_0 Duty Cycle 100ms | 1      | 0      | OK     | 12.05              |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_0 Duty Cycle 10ms  | 1      | 0      | OK     | 1.25               |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_0 Duty Cycle 30ms  | 1      | 0      | OK     | 3.65               |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_0 Frequency 100ms  | 0      | 4      | FAIL   | 10.48              |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_0 Frequency 10ms   | 1      | 0      | OK     | 10.05              |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_0 Frequency 30ms   | 1      | 0      | OK     | 3.05               |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_1 Duty Cycle 100ms | 1      | 0      | OK     | 12.06              |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_1 Duty Cycle 10ms  | 1      | 0      | OK     | 1.25               |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_1 Duty Cycle 30ms  | 1      | 0      | OK     | 3.65               |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_1 Frequency 100ms  | 0      | 4      | FAIL   | 10.48              |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_1 Frequency 10ms   | 1      | 0      | OK     | 10.05              |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_1 Frequency 30ms   | 1      | 0      | OK     | 3.05               |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_2 Duty Cycle 100ms | 1      | 0      | OK     | 12.05              |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_2 Duty Cycle 10ms  | 1      | 0      | OK     | 1.25               |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_2 Duty Cycle 30ms  | 1      | 0      | OK     | 3.65               |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_2 Frequency 100ms  | 0      | 4      | FAIL   | 10.48              |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_2 Frequency 10ms   | 1      | 0      | OK     | 10.04              |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_2 Frequency 30ms   | 1      | 0      | OK     | 3.04               |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_3 Duty Cycle 100ms | 1      | 0      | OK     | 12.04              |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_3 Duty Cycle 10ms  | 1      | 0      | OK     | 1.25               |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_3 Duty Cycle 30ms  | 1      | 0      | OK     | 3.65               |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_3 Frequency 100ms  | 0      | 4      | FAIL   | 10.48              |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_3 Frequency 10ms   | 1      | 0      | OK     | 10.04              |
| K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_3 Frequency 30ms   | 1      | 0      | OK     | 3.05               |
| K64F-GCC_ARM | K64F          | tests-api-pwm | Pwm object definable   | 1      | 0      | OK     | 0.05               |
+--------------+---------------+---------------+------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 4 FAIL / 21 OK
mbedgt: completed in 184.35 sec
mbedgt: exited with code 1
```

0xc0170, 2017-02-22:
> the K64F cannot go faster than 63.3ms, therefore the 100ms tests will always fail.

What is the limitation there?

BlackstoneEngineering, 2017-02-23:
Its a hardware limitation, as I recall their PWM register is 12bits or some such which prevents longer periods. If you go read their datasheet it states it pretty clearly. 

I consider the failure of the K64F on the 100ms test to be the CI Test shield working correctly. This is not an error on the CI Test Shield, this is the underlying hardware having an issue, one that surfaced and many people only know about after running these tests. (Eventually I want the test results displayed on the Board page and any failures to have an explanation for them)

closed 2017-10-27 by ChiefBureaucraticOfficer