ARMmbed/mbed-os #3499: K22 - CI test shield failures - i2c

0xc0170, 2016-12-22:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
K22F

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-cli version:** 
1.0.0

**meed-os sha:**
master - f6f872a

i2c tests time outs (skipped in this case).

```
mbedgt: test case report:
+----------+---------------+---------------+------------------------------------+--------+--------+---------+--------------------+
| target   | platform_name | test suite    | test case                          | passed | failed | result  | elapsed_time (sec) |
+----------+---------------+---------------+------------------------------------+--------+--------+---------+--------------------+
| K22F-ARM | K22F          | tests-api-i2c | I2C -  EEProm WR 10  Bytes         | 0      | 0      | SKIPPED | 0.0                |
| K22F-ARM | K22F          | tests-api-i2c | I2C -  EEProm WR 100 Bytes         | 0      | 0      | SKIPPED | 0.0                |
| K22F-ARM | K22F          | tests-api-i2c | I2C -  EEProm WR 2 Bytes           | 0      | 0      | SKIPPED | 0.0                |
| K22F-ARM | K22F          | tests-api-i2c | I2C -  EEProm WR Single Byte       | 0      | 0      | SKIPPED | 0.0                |
| K22F-ARM | K22F          | tests-api-i2c | I2C -  Instantiation of I2C Object | 0      | 0      | SKIPPED | 0.0                |
| K22F-ARM | K22F          | tests-api-i2c | I2C -  LM75B Temperature Read      | 0      | 0      | SKIPPED | 0.0                |
+----------+---------------+---------------+------------------------------------+--------+--------+---------+--------------------+
```

@BlackstoneEngineering @mmahadevan108

BlackstoneEngineering, 2016-12-23:
So shouldnt this be filed with NXP as a bug rather than against the CI Test shield? That is the point of the shield after all.

0xc0170, 2016-12-30:
The PR https://github.com/ARMmbed/mbed-os/pull/3502 fixes this issue, we should close it once integrated

closed 2016-12-30 by 0xc0170