ARMmbed/mbed-os #3501: NUCLEO_F401RE -  SPI, PWM and InterruptIn failues (CI test shield)

0xc0170, 2016-12-23:
### Description
- Type: Bug 
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F401RE

**Toolchain:**
GCC_ARM

**Toolchain version:**
4.9.3

**mbed-cli version:** 
1.0.0

**meed-os sha:**
 master - f6f872a

**Expected behavior**
All tests passed

**Actual behavior**
```
mbedgt: test case report:
+-----------------------+---------------+----------------------------+------------------------------------------------------------+--------+--------+---------+--------------------+
| target                | platform_name | test suite                 | test case                                                  | passed | failed | result  | elapsed_time (sec) |
+-----------------------+---------------+----------------------------+------------------------------------------------------------+--------+--------+---------+--------------------+
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin      | InterruptIn on DIO_3                                       | 0      | 0      | ERROR   | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin      | InterruptIn on DIO_4                                       | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin      | InterruptIn on DIO_5                                       | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin      | InterruptIn on DIO_6                                       | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin      | InterruptIn on DIO_7                                       | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin      | InterruptIn on DIO_8                                       | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin      | InterruptIn on DIO_9                                       | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_0 Duty Cycle 100ms                                     | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_0 Duty Cycle 10ms                                      | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_0 Duty Cycle 30ms                                      | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_0 Frequency 100ms                                      | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_0 Frequency 10ms                                       | 1      | 0      | OK      | 9.98               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_0 Frequency 30ms                                       | 0      | 0      | ERROR   | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_1 Duty Cycle 100ms                                     | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_1 Duty Cycle 10ms                                      | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_1 Duty Cycle 30ms                                      | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_1 Frequency 100ms                                      | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_1 Frequency 10ms                                       | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_1 Frequency 30ms                                       | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_2 Duty Cycle 100ms                                     | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_2 Duty Cycle 10ms                                      | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_2 Duty Cycle 30ms                                      | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_2 Frequency 100ms                                      | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_2 Frequency 10ms                                       | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_2 Frequency 30ms                                       | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_3 Duty Cycle 100ms                                     | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_3 Duty Cycle 10ms                                      | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_3 Duty Cycle 30ms                                      | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_3 Frequency 100ms                                      | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_3 Frequency 10ms                                       | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | PWM_3 Frequency 30ms                                       | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm              | Pwm object definable                                       | 1      | 0      | OK      | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-spi              | SPI - Object Definable                                     | 0      | 0      | ERROR   | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-spi              | SPI - SD Read                                              | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-spi              | SPI - SD Write                                             | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-spi              | SPI - SD card exists                                       | 0      | 0      | SKIPPED | 0.0                |
+-----------------------+---------------+----------------------------+------------------------------------------------------------+--------+--------+---------+--------------------+
```

This is out-of-box results, I haven't wired or configured anything.

@bcostm @adustm @jeromecoutant @LMESTM

bcostm, 2017-01-03:
I have tested it today:

```
+-----------------------+---------------+-----------------------+------------------------------------+--------+--------+--------+--------------------+
| target                | platform_name | test suite            | test case                          | passed | failed | result | elapsed_time (sec) |
+-----------------------+---------------+-----------------------+------------------------------------+--------+--------+--------+--------------------+
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-analogin    | Analog Input on AIN_0              | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-analogin    | Analog Input on AIN_1              | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-analogin    | Analog Input on AIN_2              | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-analogin    | Analog Input on AIN_3              | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-analogin    | Analog Input on AIN_4              | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-analogin    | Analog Input on AIN_5              | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-businout    | BusIn definable                    | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-businout    | BusIn to BusOut test               | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-businout    | BusInOut to BusInOut               | 1      | 0      | OK     | 16.06              |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-businout    | BusOut definable                   | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio   | Digital I/O on DIO_2/DIO_3         | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio   | Digital I/O on DIO_3/DIO_2         | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio   | Digital I/O on DIO_4/DIO_5         | 1      | 0      | OK     | 0.07               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio   | Digital I/O on DIO_5/DIO_4         | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio   | Digital I/O on DIO_6/DIO_7         | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio   | Digital I/O on DIO_7/DIO_6         | 1      | 0      | OK     | 0.07               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio   | Digital I/O on DIO_8/DIO_9         | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio   | Digital I/O on DIO_9/DIO_8         | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c         | I2C -  EEProm WR 10  Bytes         | 1      | 0      | OK     | 0.07               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c         | I2C -  EEProm WR 100 Bytes         | 1      | 0      | OK     | 0.08               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c         | I2C -  EEProm WR 2 Bytes           | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c         | I2C -  EEProm WR Single Byte       | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c         | I2C -  Instantiation of I2C Object | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c         | I2C -  LM75B Temperature Read      | 1      | 0      | OK     | 0.07               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin | InterruptIn on DIO_2               | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin | InterruptIn on DIO_3               | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin | InterruptIn on DIO_4               | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin | InterruptIn on DIO_5               | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin | InterruptIn on DIO_6               | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin | InterruptIn on DIO_7               | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin | InterruptIn on DIO_8               | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin | InterruptIn on DIO_9               | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm         | PWM_0 Frequency 100ms              | 1      | 0      | OK     | 10.05              |
[... cut ...]
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-spi         | SPI - Object Definable             | 0      | 0      | ERROR  | 0.0                |
+-----------------------+---------------+-----------------------+------------------------------------+--------+--------+--------+--------------------+

```
ci-test-shield version:
`c06899d` Merge pull request #14 from 0xc0170/dev_debug_msg
mbed-os version:
`e7361eb` Merge pull request #3365 from OpenNuvoton/nuvoton_usb

Note that the same ERROR are present on other platforms.  Don't know yet the root cause.

bcostm, 2017-03-03:
Retested with mbed-os-5.4

```
+-----------------------+---------------+-----------------------+------------------------------------+--------+--------+--------+--------------------+
| target                | platform_name | test suite            | test case                          | passed | failed | result | elapsed_time (sec) |
+-----------------------+---------------+-----------------------+------------------------------------+--------+--------+--------+--------------------+
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-analogin    | Analog Input on AIN_0              | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-analogin    | Analog Input on AIN_1              | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-analogin    | Analog Input on AIN_2              | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-analogin    | Analog Input on AIN_3              | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-analogin    | Analog Input on AIN_4              | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-analogin    | Analog Input on AIN_5              | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-businout    | BusIn definable                    | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-businout    | BusIn to BusOut test               | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-businout    | BusInOut to BusInOut               | 1      | 0      | OK     | 16.06              |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-businout    | BusOut definable                   | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio   | Digital I/O on DIO_2/DIO_3         | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio   | Digital I/O on DIO_3/DIO_2         | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio   | Digital I/O on DIO_4/DIO_5         | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio   | Digital I/O on DIO_5/DIO_4         | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio   | Digital I/O on DIO_6/DIO_7         | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio   | Digital I/O on DIO_7/DIO_6         | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio   | Digital I/O on DIO_8/DIO_9         | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio   | Digital I/O on DIO_9/DIO_8         | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c         | I2C -  EEProm 2nd WR 10  Bytes     | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c         | I2C -  EEProm 2nd WR 100 Bytes     | 1      | 0      | OK     | 0.09               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c         | I2C -  EEProm 2nd WR 2 Bytes       | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c         | I2C -  EEProm 2nd WR Single Byte   | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c         | I2C -  EEProm WR 10  Bytes         | 1      | 0      | OK     | 0.07               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c         | I2C -  EEProm WR 100 Bytes         | 1      | 0      | OK     | 0.08               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c         | I2C -  EEProm WR 2 Bytes           | 1      | 0      | OK     | 0.07               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c         | I2C -  EEProm WR Single Byte       | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c         | I2C -  Instantiation of I2C Object | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c         | I2C -  LM75B Temperature Read      | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin | InterruptIn on DIO_2               | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin | InterruptIn on DIO_3               | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin | InterruptIn on DIO_4               | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin | InterruptIn on DIO_5               | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin | InterruptIn on DIO_6               | 1      | 0      | OK     | 0.05               |
[... cut ...]
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm         | PWM_3 Frequency 30ms               | 1      | 0      | OK     | 3.04               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm         | Pwm object definable               | 1      | 0      | OK     | 0.06               |
+-----------------------+---------------+-----------------------+------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 61 OK
```

I think you can close this issue.

bcostm, 2017-05-18:
ST_TO_BE_CLOSED

closed 2018-01-24 by 0xc0170