ARMmbed/mbed-os #3508: NCS36510 - A4 and A5 pins are missing in PinNames

0xc0170, 2016-12-29:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NCS36510

**Actual behavior**

From ``PinNames.h`` file:

```
typedef enum {
    GPIO0 = 0,
    GPIO1,
    GPIO2,
    GPIO3,
    GPIO4,
    GPIO5,
    GPIO6,
    GPIO7,
    GPIO8,
    GPIO9,
    GPIO10,
    GPIO11,
    GPIO12,
    GPIO13,
    GPIO14,
    GPIO15,
    GPIO16,
    GPIO17,
    A0,
    A1,
    A2,
    A3,
    UART1_TX = GPIO0,
    UART1_RX = GPIO1,
    UART2_TX = GPIO8,
    UART2_RX = GPIO9,
```

Why A4 and A5 are not defined?

cc @radhika-raghavendran @pradeep-gr

radhika-raghavendran, 2017-01-02:
@0xc0170 
The base board does not have A4 and A5 connected, it is left as NC. Hence not listed in the file PinNames.h
Refer https://developer.mbed.org/platforms/NCS36510/

0xc0170, 2017-01-03:
> The base board does not have A4 and A5 connected, it is left as NC. Hence not listed in the file PinNames.h
Refer https://developer.mbed.org/platforms/NCS36510/

Thanks, a curiosity - why those pins are NC? We use those pins for https://github.com/ARMmbed/ci-test-shield tests. Here's AnalogOut test cases - https://github.com/ARMmbed/ci-test-shield/blob/master/TESTS/API/AnalogIn/AnalogIn.cpp#L67

cc @BlackstoneEngineering

BlackstoneEngineering, 2017-01-03:
If the pins are left out then the CI Test shied assertion should fail. We probably should add an assertion that tests to make sure none of the pins that are required are NC's.

maclobdell, 2017-02-07:
@0xc0170 Since the device does not have the ability to support A4 and A5, should they be set to NC in PinNames.h?  Please provide your recommendation.  Would this allow for a more graceful handling of test results when pins are not connected?
  
```
    NC          = (int)0xFFFFFFFF,
    A0,
    A1,
    A2,
    A3,
    A4 = NC,
    A5 = NC,
```

Note, I moved the definition of NC up before it is referenced.  Currently NC is defined last in the long list, which makes it difficult to use.

radhika-raghavendran, 2017-03-17:
I can see this has been added in the latest PR merge. Can you close this now?
Thanks

BlackstoneEngineering, 2017-03-17:
@0xc0170

closed 2017-03-21 by 0xc0170