ARMmbed/mbed-os #3512: Ethernet Multicast

sstaub, 2016-12-31:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
LPC1768

The UDPSocket class member
`int join_multicast_group(const char *address) `
is missing in mbed-os 5, please bring it back.

0xc0170, 2017-01-03:
cc @geky @c1728p9

sstaub, 2017-07-19:
Are there any news on multicast support, or can I put my mbed boards into the trash.

geky, 2017-07-19:
You can send them to me if you don't want them ; )

But in seriousness, no, we do not have an update on multicast support.

There is no technical reason we haven't been able to reintroduce multicast support, we simply haven't had the time to get to it yet. It's unfortunately not as simple as copying over the code from mbed 2, since we will need to set up a multicast network to test and make sure multicast is still working with the LWIP changes we've made internally. We also have several other outstanding network issues (cough emac api cough) that are in progress and are being considered as higher priority.

Unfortunately, the best I can leave you with right now is pointers if you're interested in bringing over the multicast support yourself:

- Add the setsockopt flags `NSAPI_ADD_MEMBERSHIP` and `NSAPI_DROP_MEMBERSHIP` to the nsapi:
  https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/nsapi_types.h#L201-L217
- Add handling for `NSAPI_ADD_MEMBERSHIP` and `NSAPI_DROP_MEMBERSHIP` to lwip's setsockopt function:
  https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_LWIP/lwip-interface/lwip_stack.c#L983
- The code from the mbed 2 sockets should be a good starting point:
  https://developer.mbed.org/users/mbed_official/code/Socket/docs/434906b5b977/UDPSocket_8cpp_source.html#l00052

If anyone else is able to take care of this, feel free to pick it up.

SeppoTakalo, 2018-06-14:
This is already in: https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/Socket.h#L70

So this issue should be closed.

closed 2018-06-15 by 0xc0170