ARMmbed/mbed-os #3515: UBLOX_C027: Blinky example not working with new mbed-os

melomaa, 2017-01-03:
### Description
- Type: Bug
- Priority: Major

**Target**
UBLOX_C027

**Toolchain:**
GCC_ARM

**Toolchain version:**
Tested with 4.9.3 and 5.4.1

**mbed-cli version:** 
1.0.0

**mbed-os sha:**
0789928 (mbed-os-5.3) and d5de476 (mbed-os-5.2)

**Expected behavior**
LED blinks

**Actual behavior**
LED does not blink

**Steps to reproduce**
Compile mbed-os-example-blinky
mbed compile -t GCC_ARM -m UBLOX_C027
Copy binary to target and reset

**Notes**
Works with mbed-os 5.1 (a6f3fd1)

0xc0170, 2017-01-03:
@mazgch

ashok-rao, 2017-01-27:
@mazgch .. Could you please check this? Thanks.

ashok-rao, 2017-02-07:
@mazgch @0xc0170 .. Any progress on this? @melomaa has been wanting to test out the hardware with mbed OS 5.3 asap..

0xc0170, 2017-02-07:
I don't have the hw at the moment. Could you @ashok-rao get a hw to test this?

@andreaslarssonublox - Could you pinpoint anyone to help with this?

ashok-rao, 2017-02-07:
@0xc0170 .. I don't seem to have this board as well. I guess we'll have to wait for @mazgch. Thanks!

RobMeades, 2017-03-23:
Yup, well found @andreaslarssonublox, seems likely. See pull request #3990.

0xc0170, 2017-04-25:
The PR previously referenced fixed this

closed 2017-04-25 by 0xc0170