ARMmbed/mbed-os #3516: Macros aren't passed to TESTS folder during "mbed test" command

jenia81, 2017-01-03:
### Description
- Type: Bug 
- Priority:  Major 

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM|ARM

**mbed-cli version:** 
1.0.0

**Expected behavior**
macros that are passed to the "mbed test" command should be passed to TESTS files too.

**Actual behavior**
macros that are passed to the "mbed test" command aren't passed to TESTS files.

From looking into the code of test_api.py script, it seems that the macros that are passed by the user during "mbed test" command execution are overwritten by another function.

From test_api.py:
def build_tests(... ,macros=None, ...) {
...
cfg, macros, features = get_config(base_source_paths, target_name, toolchain_name)
.. )
}

0xc0170, 2017-01-03:
cc @theotherjimmy @bridadan

bridadan, 2017-01-03:
Thanks for raising this issue @jenia81. The line you mentioned was indeed the problem, I've submitted PR #3518 to fix this. Would you mind verifying that my changes fix the issue?

jenia81, 2017-01-04:
@bridadan No problem, I'll check and let you know

bridadan, 2017-01-05:
@jenia81 Did you have a chance to check it out?

jenia81, 2017-01-05:
@bridadan I've checked and it's working

PR #3518 (2017-01-09): Preventing app_config from clobbering CLI macros
## Description
This fixes #3516. If an `mbed_app.json` file is used when compiling tests, the tools were incorrectly ignoring macros that were supplied on the command line. This PR fixes this so the macros propagate correctly.

## Status
**READY**

## Migrations
If this PR changes any APIs or behaviors, give a short description of what *API users* should do when this PR is merged.

NO

## Todos
- [ ] Tests

## Steps to test or reproduce
Edit any test with the following code:

```c++
#ifdef MY_SILLY_MACRO
printf("MY_SILLY_MACRO is set!\r\n");
#el [... cut]
closed 2017-01-09 by sg-