ARMmbed/mbed-os #3520: STM32L4 mbed start resets the RTC to zero.

helmut64, 2017-01-03:
- Type: Bug STM32L4 (all STM boards)
I figured out that every time a CPU reset gets done the RTC clock gets inititalized to zero. I believe I figured out where: lp_ticker_init() -> rtc_init() ->rtc_write(0);

I feel that this is a bug because the power is still active and my RTC clocks continues. It should not reset the clock unless there is an invalid dated during startup, e.g. less than the current mbed build time -1 day.

It is easy to reproduce:
printf("rtc_read: %x", rtc_read());
If you set the RTC via set_time(value) the correct time works, however after a soft or hard reset the RTC is zero.

bridadan, 2017-01-03:
cc @adustm @bcostm @jeromecoutant

helmut64, 2017-01-06:
I am sorry, no bug in mbed, it has been a software but in my app. I close this.

closed 2017-01-06 by helmut64