ARMmbed/mbed-os #3523: NUCLEO 103RB: Why removing flag DEVICE_SERIAL_FC=1 gives this error?

peeyushpashine, 2017-01-04:
When removing DEVICE_SERIAL_FC=1 from Nucleo F103RB gives the below linking error while compiling:
BufferedSerial.o:(.rodata._ZTV14BufferedSerial+0x4c): undefined reference to `non-virtual thunk to mbed::Serial::lock()' 
BufferedSerial.o:(.rodata._ZTV14BufferedSerial+0x50): undefined reference to `non-virtual thunk to mbed::Serial::unlock()' 
BufferedSerial.o:(.rodata._ZTV14BufferedSerial+0x54): undefined reference to `non-virtual thunk to mbed::Serial::_putc(int)' 
BufferedSerial.o:(.rodata._ZTV14BufferedSerial+0x58): undefined reference to `non-virtual thunk to mbed::Serial::_getc()' 

I understand the this error and target.json for Nucleo 103RB has DEVICE_SERIAL_FC flag. However for me removing this flag in case of other hardware does not give any compiling error. I went through the source code serial_api.c for all hardwares and it looks similar. where is the difference occurring for 103RB?

LMESTM, 2017-01-04:
Have you tried to clean (-c option) and even to manually remove build folders ?

peeyushpashine, 2017-01-05:
Yes I did that. cleaning the project, deleting the build folders and re-building it as well gives the same issue. I do not find this virtual methods _putc(), getc() defined under the DEVICE_SERIAL_FC macro and which is the case with other hardware as well, still only for 103rb i face this issue.

bcostm, 2017-01-05:
No I have no explanation.

I compiled sucessfully the NUCLEO_F103RB with DEVICE_SERIAL_FC removed from targets.json.

I checked the following command lines:
python build.py -v -x -c -m NUCLEO_F103RB -t ARM -j 4
python build.py -v -x -c -m NUCLEO_F103RB -t GCC_ARM -j 4
mbed test -vv -m NUCLEO_F103RB -t ARM --compile
mbed test -vv -m NUCLEO_F103RB -t GCC_ARM --compile

What is your command line to compare ?

sg-, 2017-02-07:
@peeyushpashine Bump!

closed 2017-10-27 by ChiefBureaucraticOfficer