ARMmbed/mbed-os #3524: EthernetInterface MBED-OS 5.3 static IP

andcor02, 2017-01-04:
Hi,

I am forwarding the issue from multiple users have problems setting static IP in MBED OS 5.

**Problem:**

- Users cannot set static IP

- Users have tried to use: `virtual nsapi_error_t set_network(const char *ip_address, const char *netmask, const char *gateway);`

However the following errors are output

Example network-socket TCP Server
set IP status: 0
connect status: 0
IP address is: 192.168.1.100
MAC address is: 8a:42:7d:87:0c:b7
open error: -3004

Error -3004 is "not connected to a network"
However I can ping the module on it's assigned IP address and it responds to the ping so we know the network is indeed connected.

Link to original message:

https://developer.mbed.org/questions/76349/EthernetInterface-MBED-OS-53-static-IP/#answer11746

Regards,

Andrea

bridadan, 2017-01-04:
cc @geky

geky, 2017-01-04:
Related issue https://github.com/ARMmbed/mbed-os/issues/3234#issuecomment-261019778

It looks like the initialization of lwip for static ip addresses doesn't set a dns server. It does for ipv6, but it looks like the path for ipv4 relies on dns servers from dhcp. Working on a patch to fix this.

geky, 2017-01-04:
Hopefully #3526 resolves all these problems, let me know if there are still issues.

sextondw, 2017-02-27:
This issue still seems to exist. I have tried to use 5.3.3 and 5.3.5 and still have no success setting a static IP address. The original ticket is: https://github.com/ARMmbed/mbed-os/issues/3524

I believe this is still an issue.

cldpinheiro, 2017-07-19:
Unfortunately it is not working yet.

andcor02, 2017-07-26:
@geky 

Do we have a time to know when this will bed fixed?

Regards,

Andrea

kegilbert, 2017-07-27:
@cldpinheiro what device and compiler did you test with? On a K64F (mentioned as the device used in the [original issue](https://developer.mbed.org/questions/76349/EthernetInterface-MBED-OS-53-static-IP/#answer11746)) with GCC_ARM at the head of mbed-os (commit 78a1eddf61efa705ef1e99fc597dcaa5e737fb4c) I cannot reproduce the error. I get the following running the linked example:

```
Example network-socket TCP Server 
set IP status: 0                  
connect status: 0                 
IP address is: 192.168.1.100      
MAC address is: a6:43:49:7a:c2:c2 
server open                       
Bound                             
Listening             
```

0xc0170, 2018-06-14:
Due to inactivity (no update how to reproduce this issue), closing

closed 2018-06-14 by 0xc0170