ARMmbed/mbed-os #3525: Tests: spawned processes not killed when 'mbed test' terminated

bulislaw, 2017-01-04:
### Description
- Type: Bug
- Priority: Minor

## Bug

`mbed test [...]` spawns number of processes that keep running after the main application is terminated (eg using Ctrl+c), requiring user to find them and kill them manually when they want to cancel the test run.

We should be trapping the KeyboardInterrupt and killing the processes we spawned cleanly.

@bridadan @mazimkhan

bridadan, 2017-01-04:
Thanks for reporting this @bulislaw, can you give me more details about your environment so I can reproduce it?

Specifically, could you provide your:
- OS
- Python version
- output of `pip list`
- The output of `mbed test [...]` when you hit Ctrl-C

Thanks!

bulislaw, 2017-01-05:
I wouldn't think it's setup specific.
OS: MacOS and Linux
Python: 2.7.12
`pip list`
```
appnope (0.1.0)
backports.shutil-get-terminal-size (1.0.0)
beautifulsoup4 (4.5.1)
cmsis-svd (0.4)
colorama (0.3.7)
decorator (4.0.10)
enum34 (1.1.6)
esptool (1.1)
fasteners (0.14.1)
funcsigs (1.0.2)
functools32 (3.2.3.post2)
future (0.15.2)
fuzzywuzzy (0.12.0)
hidapi (0.7.99.post19)
intelhex (2.1)
ipython (5.1.0)
ipython-genutils (0.1.0)
Jinja2 (2.8)
jsonschema (2.5.1)
junit-xml (1.7)
lockfile (0.12.2)
MarkupSafe (0.23)
mbed-cli (1.0.0, /usr/local/lib/python2.7/site-packages)
mbed-greentea (1.2.4)
mbed-host-tests (1.1.5)
mbed-ls (1.2.10)
mechanize (0.2.5)
mercurial (3.9.2)
mock (2.0.0)
monotonic (1.2)
netifaces (0.10.5)
pathlib2 (2.1.0)
pbr (1.10.0)
pexpect (4.2.1)
pickleshare (0.7.4)
pip (9.0.1)
Powerline-font-patcher (beta)
prettytable (0.7.2)
project-generator (0.9.10)
project-generator-definitions (0.2.33)
prompt-toolkit (1.0.8)
protobuf (3.1.0.post1)
ptyprocess (0.5.1)
pyasn1 (0.1.9)
pyasn1-modules (0.0.8)
pycurl (7.43.0)
Pygments (2.1.3)
pyOCD (0.8.0, /Users/barsza01/devel/mbed/tools/pyOCD)
pyserial (3.2.1)
python-Levenshtein (0.12.0)
pyxdg (0.25)
PyYAML (3.12)
requests (2.12.3)
setuptools (28.6.1)
simplegeneric (0.8.1)
six (1.10.0)
traitlets (4.3.1)
urlgrabber (3.9.1)
vboxapi (1.0)
wcwidth (0.1.7)
websocket-client (0.37.0)
wheel (0.29.0)
xmltodict (0.10.2)
```

Output:
I'm getting `[mbed] User aborted!` but the background processes keep going 
```
[1483613010.45][HTST][INF] No events in queue
[1483613010.45][HTST][INF] stopped consuming events
[1483613010.45][HTST][INF] host test result() call skipped, received: True
[1483613010.45][HTST][INF] calling blocking teardown()
[1483613010.45][HTST][INF] teardown() finished
[1483613010.46][HTST][INF] {{result;success}}
^Cmbedgt: checking for GCOV data...
[mbed] User aborted!
mbedgt: mbed-host-test-runner: stopped and returned 'OK'
mbedgt: test on hardware with target id: 024002265f2e1e4d000000000000000000000000a2f7e3f5
mbedgt: test suite 'features-frameworks-utest-tests-unit_tests-test_setup_failure' ................... OK in 12.19 sec
	test case: 'dummy test' ...................................................................... OK in 0.03 sec
	test case: 'dummy test 2' .................................................................... OK in 0.04 sec
mbedgt: test case summary: 2 passes, 0 failures
mbedgt: checking for 'host_tests' directory above image directory structure
	found 'host_tests' directory in: 'TESTS/host_tests'
mbedgt: selecting test case observer...
	calling mbedhtrun: mbedhtrun -m K64F -p /dev/tty.usbmodem1412:9600 -f "BUILD/tests/k64f/GCC_ARM/TESTS/mbedmicro-rtos-mbed/queue/queue.bin" -d /Volumes/MBED -C 4 -c shell -t 024002265f2e1e4d000000000000000000000000a2f7e3f5 -e "TESTS/host_tests"
mbedgt: mbed-host-test-runner: started
[1483613010.85][HTST][INF] host test executor ver. 1.1.5                                                                                                                                                    
[1483613010.85][HTST][INF] copy image onto target...
[1483613010.85][COPY][INF] Waiting up to 60 sec for '024002265f2e1e4d000000000000000000000000a2f7e3f5' mount point (current is '/Volumes/MBED')...
 ✘  ~/devel/mbed/code/mbed-os   master  [1483613019.72][HTST][INF] starting host test process...
[1483613019.72][CONN][INF] starting connection process...
[1483613019.72][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1483613019.72][CONN][INF] initializing serial port listener...
[1483613019.72][SERI][INF] serial(port=/dev/tty.usbmodem1412, baudrate=9600, timeout=0.01)
[1483613019.72][PLGN][INF] Waiting up to 60 sec for '024002265f2e1e4d000000000000000000000000a2f7e3f5' serial port (current is '/dev/tty.usbmodem1412')...
[1483613019.73][HTST][INF] setting timeout to: 60 sec
[1483613020.20][SERI][INF] reset device using 'default' plugin...
[1483613020.60][SERI][INF] waiting 1.00 sec after reset
[1483613021.60][SERI][INF] wait for it...
[1483613021.60][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1483613021.60][CONN][INF] sending up to 2 __sync packets (specified with --sync=2)
[1483613021.60][CONN][INF] sending preamble '1332b08d-b72a-4a08-b774-9e51f4d3b71e'
[1483613021.60][SERI][TXD] {{__sync;1332b08d-b72a-4a08-b774-9e51f4d3b71e}}
[1483613021.73][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1483613021.78][CONN][INF] found SYNC in stream: {{__sync;1332b08d-b72a-4a08-b774-9e51f4d3b71e}} it is #0 sent, queued...
[1483613021.78][HTST][INF] sync KV found, uuid=1332b08d-b72a-4a08-b774-9e51f4d3b71e, timestamp=1483613021.781833
[1483613021.80][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1483613021.80][HTST][INF] DUT greentea-client version: 1.3.0
[1483613021.83][CONN][INF] found KV pair in stream: {{__timeout;20}}, queued...
[1483613021.83][HTST][INF] setting timeout to: 20 sec
[1483613021.86][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1483613021.86][HTST][INF] host test class: '<class 'mbed_host_tests.host
```

bulislaw, 2017-01-05:
Right now i'm using `ps aux | grep mbed-os | awk '{print $2}' | xargs kill` to kill the tests after i terminate the main process with Ctrl-C

bridadan, 2017-01-05:
I believe this is a problem that only affects MacOS and Linux. I can't reproduce the issue on Windows machines. @geky you see this issues as well, correct?

geky, 2017-01-06:
Yes, seems to pop up on *nux machines. I think Windows is more aggressive at terminating child processes, so that's probably why this issue isn't apparent in Windows.

closed 2017-10-27 by ChiefBureaucraticOfficer