ARMmbed/mbed-os #3527: LPC17XX CAN::attach OS 5.2

pilotak, 2017-01-04:
When I need to attach RxIrq on CAN, program runs fine until I start transmitting data on bus and than the program freezes, only hardware reset button helps. Is there a bug or am I missing something? it works with the old mbed (not OS)```http://mbed.org/users/mbed_official/code/mbed/builds/63bcd7ba4912```

```cpp
CAN can(P0_4, P0_5); 

void canread(){
    CANMessage msg; 
    can.read(msg);
}

int main() {
    can.frequency(250000);
    can.reset();
    can.attach(&canread, CAN::RxIrq);
    // can.attach(callback(&canread), CAN::RxIrq); // tried too
    // can.attach(callback(canread), CAN::RxIrq); // tried too
    // can.attach(canread, CAN::RxIrq); // tried too
}
```

### Description
- Type: Bug | Question
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
LCP17xx

**Toolchain:**
mBed IDE online

**OS version:** 
`revison 2885c1b`

**Expected behavior**
to work as i should

**Actual behavior**
stops whole program when transmitting data and CAN::attach is presented

0xc0170, 2017-01-05:
cc @toyowata

pilotak, 2017-01-05:
Another note, when i tried to put it into event queue i get an error through serial port:
`RTX error code: 0x00000002, task ID: 0x00000000` if that helps
```cpp
#include "mbed.h"
#include "mbed_events.h"
EventQueue queue(32 * EVENTS_EVENT_SIZE);

CAN can(P0_4, P0_5); 
uint32_t i = 0;

void canread(){
    CANMessage msg; 
    can.read(msg);
    printf("Received: %u\r\n", msg.id);
}

int main() {
    can.frequency(250000);
    can.reset();
    can.attach(queue.event(canread), CAN::RxIrq);
    
    while (true) {
        Thread::wait(500);
        printf("alive %u\r\n", i);
        i++;
    }
}
```

vargham, 2017-01-30:
RTX error code: 0x00000002 is FIFO overflow. You have to read the CAN message in the interrupt service routine to clear interrupt flag. The event handler is NOT the ISR.

pilotak, 2017-02-02:
I did that too, please read my [opening post](#issue-198794877)

vargham, 2017-02-02:
Code in opening post is different. That is a super loop type program.
Second one uses RTOS (Real Time Operating System). MBED uses Keil RTX RTOS in the background. MBED events are based on RTX features, so MBED included that for you in the background. RTX error message 0x00000002 is from RTX kernel.
`can.attach(queue.event(canread), CAN::RxIrq);` automagically attaches Interrupt Service Routine (ISR) which passes forward as an event (signals another thread), so your 'void canread()' does not run in interrupt context (actually never runs). You have to read can message in interrupt context to clear interrupt flag. Without clearing the flag, the irq immediately fires again when ISR exits, so ISR runs again till thread signal's FIFO overflows.

vargham, 2017-02-02:
I tried the code in your first post with LPC1768 (MBED release 135). Works fine.

pilotak, 2017-05-04:
I can confirm that opening post now works on mBed-OS 5.4.4. If someone finds this post: as it's an ISR you can't use ie. `printf()` and variables should be `volatile`

Thanks

closed 2017-05-04 by pilotak