ARMmbed/mbed-os #3529: Eclipse export generates wrong formatted Include Paths inside the .cproject

bittailor, 2017-01-05:
### Description
- Type: Bug
- Priority:  Minor

When exporting a application to eclipse `mbed export -i eclipse_gcc_arm` and importing it into an Eclipse workspace Eclipse shows lots of `Invalid project path:` warnings:

[image]

This is because the include paths are wrong formatted for eclipse. They are `/./<path>` instead of `/<project>/<path>`:

[image]

I locally fixed the generator to format it correctly (will do a pull request) and the warnings are gone:

[image]

and the paths look like this:

[image]

---------------------------------------------------------------
## Bug

**Target**
-

**Toolchain:**
GCC_ARM

**Toolchain version:**
-

**mbed-cli version:** 
1.0.0

**meed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**
-

**Expected behavior**
correct formatted Eclipse include paths

**Actual behavior**
wrong formatted Eclipse include paths

**Steps to reproduce**
- exporting an application to eclipse `mbed export -i eclipse_gcc_arm`
- importing it into an Eclipse workspace

0xc0170, 2017-01-05:
cc @theotherjimmy

bittailor, 2017-01-05:
See pull request #3531

janjongboom, 2017-01-05:
@bittailor Which OS is this on? I don't have this issue on macOS 10.11.2...

bittailor, 2017-01-05:
OS is **OS X El Capitan 10.11.6** so nearly the same as yours.
Eclipse is **Neon Release (4.6.0)**

bittailor, 2017-01-05:
@janjongboom do you also have no warning signs (orange triangle with exclamation mark) on the include directories when you open `Properties` with a right click on the mbed project and then go to  `C++ General => Paths and Symbols` and then select `Includes` tab?

janjongboom, 2017-01-06:
@bittailor Ah there, yes, I see that. Good catch. 👍

theotherjimmy, 2017-01-12:
@bittailor, your PR was merged. Can this issue be closed?

bittailor, 2017-01-12:
Fixed with PR #3531

closed 2017-01-12 by bittailor