ARMmbed/mbed-os #3533: Eclipse pyOCD debugging does not work on macOS / Linux out of the box due to hard-coded .exe

janjongboom, 2017-01-05:
In pyocd_settings.html we define:

```
<stringAttribute key="org.eclipse.cdt.dsf.gdb.DEBUG_NAME" value="arm-none-eabi-gdb.exe"/>
```

Which is obviously not going to work on non-Windows systems. I'm wondering how we could fix that. Just pointing to `arm-none-eabi-gdb` would be enough, the makefile depends on the PATH anyway.

@theotherjimmy

theotherjimmy, 2017-01-05:
*sigh* I thought @sarahmarshy changed that to exclude the `.exe` suffix, but apparently it did not get in. 

You are correct: This is clearly wrong. PRs definitely accepted.

janjongboom, 2017-01-05:
@theotherjimmy So... the variable is in the pyOCD plugin for Eclipse [here](https://github.com/gnuarmeclipse/plug-ins/blob/develop/ilg.gnuarmeclipse.debug.gdbjtag.pyocd/preferences.ini#L60) but no idea how to expose this to the Eclipse variables...

ChiefBureaucraticOfficer, 2017-10-27:
GitHib issue review: Closed due to inactivity.  Please re-file if critical issues found.

closed 2017-10-27 by ChiefBureaucraticOfficer

janjongboom, 2017-10-30:
@ChiefBureaucraticOfficer This is still an issue.

sarahmarshy, 2017-10-30:
@janjongboom, it looks like a fix was submitted- https://github.com/ARMmbed/mbed-os/commit/4b5535f9b6d44c7dcf4a63f24ef59a76b55aed25

theotherjimmy, 2017-10-30:
@janjongboom  see here: https://github.com/ARMmbed/mbed-os/blob/master/tools/export/cdt/pyocd_settings.tmpl#L52 there is no `.exe` suffix.

janjongboom, 2017-10-30:
OK, if this works then we should close it.