ARMmbed/mbed-os #3542: STM32L4 deepsleep broken between 132 and 133 by Jerome Countant changes

helmut64, 2017-01-07:
The STM32L4 deeplseep in rev-133 needs 120uA instead of 7uA in rev-132 due to the changes Jerome Countant in checkin: 8627204d3667dfc548c2b138d70f0bb065fdd0d0, see changes in sleep.c->deepsleep, restoring this changes is a workaound. I believe the "HAL_PWREx_EnableLowPowerRunMode()" enable/disable is missing.
- Type: Bug 

I tested this with the L4 Nucleo64 board.

BTW: If there is more we can do to get the power consumption down the a few more uA I would be interested in this. Accounting to the data sheet Stop2 with RTC should eat only 2uA. Any advise would be appreciated.

Regards Helmut

jeromecoutant, 2017-01-09:
Hi
I confirm that current deepsleep consumption is 120uA with NUCLEO_L476RG
This becomes back to 7uA after adding HAL_PWREx_EnableLowPowerRunMode() call

Thanks Helmut for issue reporting

Jerome

helmut64, 2017-01-22:
The problem is solved and I verified it.

closed 2017-01-22 by helmut64