ARMmbed/mbed-os #3543: mbed-events breaks Timeout on STM32F4

DavidPowell, 2017-01-08:
### Description
- Type: Bug 
- Related issue: ARMmbed/mbed-events#16
- Priority: Major 
---------------------------------------------------------------
## Bug

**Target**
STM32F4

**Toolchain:**
GCC_ARM

**mbed version:** 
Revision 131 (bundled with platformio)

**Expected behavior**

Using `Timeout` to creating a callback to turn off a pin after 50us, I expect to see the pin turn off when looking with an oscilloscope. I am create an `EventQueue` from mbed-events, which is not expected to interact with the `Timeout`.

**Actual behavior**

On NUCLEO L053R8: The callback turns off the pin after 78us
On NUCLEO F401RE , the pin goes high, and stays high. Debugging with GDB indicates that the callback never gets called. (same behaviour on Discovery F4 after changing `my_pin` to a valid pin for this board)

Replacing the call to `queue.dispatch()` with `while (1) {}` means that the callback is now called properly on both targets. This suggests that some code called by the queue is blocking interrupts.

The issue has some similarities to ARMmbed/mbed-events#16, where the `sleep()` function in the HAL is a suspect. Since the behaviour depends on the target, I have raised the issue here, where all target dependent code lives

**Steps to reproduce**

The following example code illustrates the problem:
````c++
#include "mbed.h"
#include "mbed_events.h"

DigitalOut my_pin(D2);

void pin_off(void)
{
  my_pin = 0;
}

EventQueue queue;

int main(void)
{
  printf("starting\n");
  Timeout to;
  while (1)
  {
    my_pin = 1;
    to.attach_us(pin_off, 50);
    //while(1) {};
    queue.dispatch();
  }
}
````

0xc0170, 2017-01-09:
cc @bcostm @adustm @LMESTM @jeromecoutant

bridadan, 2017-01-09:
May also be of interest to @geky

geky, 2017-01-10:
Hi @DavidPowell, thanks for raising the issue and example. Looks like this was a simple bug in the non-rtos implementation of mbed-events.

Interestingly, it wasn't an issue with critical sections, but actually an unchecked overflow when the event queue tries to create a timeout before going to sleep. This line didn't check for -1, which is used in mbed-events to represent an infinite wait/dispatch:
https://github.com/ARMmbed/mbed-os/blob/master/events/equeue/equeue_mbed.cpp#L127

The result was a timeout with some ridiculously large delay, which would mostly work on platforms (for ~50 days). Oddly enough it seemed to kill any other timeouts on the STM32F4 boards.

Raised a pr here: https://github.com/ARMmbed/mbed-os/pull/3550, let me know if that doesn't fix the problem.

DavidPowell, 2017-01-10:
@geky Thanks for the fix. I applied it manually, and confirmed that it solves both the problem both in my example code above, and in my real code.

PR #3550 (2017-01-12): events - Fix overflow of timeout on STM32F4
For equeue_sema_wait, -1 is used to indicate an infinite wait. This wasn't handled in the nonrtos implementation and caused undefined/weird behaviour after an overflow on integer multiplication.

On most boards, the infinite wait would return after ~50 days, on the STM32F4 the timeout killed all other timeouts for some reason.

tested on NUCLEO_F401RE
should resolve #3543
mirrored on https://github.com/ARMmbed/mbed-events/pull/27
cc @DavidPowell
closed 2017-01-12 by adbridge