ARMmbed/mbed-os #3549: [ARCH_PRO] features-feature_lwip-tests-mbedmicro-net-udp_dtls_handshake test failing

bridadan, 2017-01-09:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
ARCH_PRO

**Toolchain:**
GCC_ARM, ARM, and IAR

**mbed-cli version:** 
1.0.0

**mbed-os sha:**
3a326c0b9411c16b44ccca47c5deb2edf340642b

**Expected behavior**
Test should pass.

**Actual behavior**
The test does not pass! Below is the output from the test:

```
mbedgt: checking for 'host_tests' directory above image directory structure
        found 'host_tests' directory in: 'features\FEATURE_LWIP\TESTS\mbedmicro-net\host_tests'
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m ARCH_PRO -p COM334:9600 -f "BUILD/tests/ARCH_PRO/GCC_ARM/features/FEATURE_LWIP/TESTS/mbedmicro-net/udp_dtls_handshake/udp_dtls_handshake.bin" -d E: -C 4 -c shel
l -t 90040000e48dd08e00000000000000000000000097969902 -e "features\FEATURE_LWIP\TESTS\mbedmicro-net\host_tests"
mbedgt: mbed-host-test-runner: started
[1483990246.14][HTST][INF] host test executor ver. 1.1.6
[1483990246.14][HTST][INF] copy image onto target...
[1483990246.14][COPY][INF] Waiting up to 60 sec for '90040000e48dd08e00000000000000000000000097969902' mount point (current is 'E:')...
        1 file(s) copied.
[1483990254.95][HTST][INF] starting host test process...
[1483990255.33][CONN][INF] starting connection process...
[1483990255.33][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1483990255.33][CONN][INF] initializing serial port listener...
[1483990255.33][PLGN][INF] Waiting up to 60 sec for '90040000e48dd08e00000000000000000000000097969902' serial port (current is 'COM334')...
[1483990255.34][HTST][INF] setting timeout to: 60 sec
[1483990255.49][SERI][INF] serial(port=COM334, baudrate=9600, timeout=0.01)
[1483990255.49][SERI][INF] reset device using 'default' plugin...
[1483990255.75][SERI][INF] waiting 1.00 sec after reset
[1483990256.75][SERI][INF] wait for it...
[1483990256.75][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1483990256.75][CONN][INF] sending up to 2 __sync packets (specified with --sync=2)
[1483990256.75][CONN][INF] sending preamble 'a8045e1f-ebd0-4c60-b84c-89f94728138a'
[1483990256.75][SERI][TXD] {{__sync;a8045e1f-ebd0-4c60-b84c-89f94728138a}}
[1483990256.88][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1483990256.93][CONN][INF] found SYNC in stream: {{__sync;a8045e1f-ebd0-4c60-b84c-89f94728138a}} it is #0 sent, queued...
[1483990256.93][HTST][INF] sync KV found, uuid=a8045e1f-ebd0-4c60-b84c-89f94728138a, timestamp=1483990256.927000
[1483990256.95][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1483990256.95][HTST][INF] DUT greentea-client version: 1.3.0
[1483990256.97][CONN][INF] found KV pair in stream: {{__timeout;20}}, queued...
[1483990256.97][HTST][INF] setting timeout to: 20 sec
[1483990257.01][CONN][INF] found KV pair in stream: {{__host_test_name;udp_shotgun}}, queued...
[1483990257.02][HTST][INF] host test class: '<class 'udp_shotgun.UDPEchoClientTest'>'
[1483990257.02][HTST][INF] host test setup() call...
[1483990257.02][HTST][INF] CALLBACKs updated
[1483990257.02][HTST][INF] host test detected: udp_shotgun
[1483990266.04][CONN][RXD] MBED: UDPClient IP address is '192.168.1.94'
[1483990266.09][CONN][RXD] MBED: UDPClient waiting for server IP and port...
[1483990266.12][CONN][INF] found KV pair in stream: {{target_ip;192.168.1.94}}, queued...
[1483990266.13][HTST][INF] HOST: Listening for UDP packets: 10.118.14.89:64737
[1483990266.14][CONN][INF] found KV pair in stream: {{host_ip; }}, queued...
[1483990266.15][SERI][TXD] {{host_ip;10.118.14.89}}
[1483990266.20][CONN][INF] found KV pair in stream: {{host_port; }}, queued...
[1483990266.21][SERI][TXD] {{host_port;64737}}
[1483990266.30][CONN][RXD] MBED: UDP Server IP address received: 10.118.14.89:64737
[1483990266.34][CONN][RXD] MBED: DTLS pattern [128, 400, 208, 224, 32]
[1483990266.36][CONN][RXD] UDP: tx -> 5
[1483990266.39][CONN][RXD] UDP: rx <- 32 x (expected 128)
[1483990266.41][CONN][RXD] UDP: tx -> 5
[1483990266.42][CONN][RXD] UDP: rx <- 128
[1483990266.44][CONN][RXD] UDP: rx <- 400
[1483990277.59][HTST][INF] test suite run finished after 20.62 sec...
[1483990277.59][CONN][INF] received special even '__host_test_finished' value='True', finishing
[1483990277.60][HTST][INF] CONN exited with code: 0
[1483990277.60][HTST][INF] No events in queue
[1483990277.60][HTST][INF] stopped consuming events
[1483990277.60][HTST][INF] host test result(): None
[1483990277.60][HTST][WRN] missing __exit event from DUT
[1483990277.60][HTST][WRN] missing __exit_event_queue event from host test
[1483990277.60][HTST][ERR] missing __exit_event_queue event from host test and no result from host test, timeout...
[1483990277.60][HTST][INF] calling blocking teardown()
[1483990277.88][HTST][INF] teardown() finished
[1483990277.88][HTST][INF] {{result;timeout}}
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'TIMEOUT'
mbedgt: test case summary event not found
        no test case report present, assuming test suite to be a single test case!
        test suite: features-feature_lwip-tests-mbedmicro-net-udp_dtls_handshake
        test case: features-feature_lwip-tests-mbedmicro-net-udp_dtls_handshake
mbedgt: test on hardware with target id: 90040000e48dd08e00000000000000000000000097969902
mbedgt: test suite 'features-feature_lwip-tests-mbedmicro-net-udp_dtls_handshake' .................... TIMEOUT in 32.16 sec
        test case: 'features-feature_lwip-tests-mbedmicro-net-udp_dtls_handshake' .................... ERROR in 32.16 sec
mbedgt: test case summary: 0 passes, 1 failure
```

**Steps to reproduce**
```
mbed test -m ARCH_PRO -t GCC_ARM -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 --compile --clean -n features-feature_lwip-tests-mbedmicro-net-udp_dtls_handshake
mbed test -m ARCH_PRO -t GCC_ARM --run -n features-feature_lwip-tests-mbedmicro-net-udp_dtls_handshake -v
```

c @geky

MarceloSalazar, 2017-01-23:
I can also replicate this...
It would really help us to get this working asap ;)

geky, 2017-02-15:
Sorry about the delay, #3781 should fix this.

closed 2017-02-17 by sg-