ARMmbed/mbed-os #3553: IDE projects - folders and files to be sorted alphabetically

0xc0170, 2017-01-10:
### Description
- Type: Enhancement 
- Priority:  Minor

## Enhancement

**Suggested enhancement**

It's time consuming to explore a project in a workspace view (I am currently using uvision) as files and folders are not sorted, therefore it takes time to find a folder/file.

This is one project from current master:
[image]

cc @theotherjimmy @sarahmarshy

theotherjimmy, 2017-01-10:
I have mocked it up for Uvision. I can do something similar for IAR. Which IDEs are we going to do this with?

0xc0170, 2017-01-11:
> I have mocked it up for Uvision. I can do something similar for IAR. Which IDEs are we going to do this with?

I would have it as default for any tool that provides workspace view (or even for all, having this as default behavior).

PR #3562 (2017-01-12): Alphabetize UVision groups
## Status
**READY**

Might resolve #3553

## Todos
- [x] Tests /morph export-build
closed 2017-01-12 by adbridge